OrcaSlicer/OrcaSlicer #9048: Error uploading to print host

cleighland, 2025-03-23:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0

### Operating System (OS)

macOS

### OS Version

MacOS 15.2 (ARM)

### Additional system information

CPU: Apple M2
RAM: 8GB

### Printer

Custom CoreXY running Klipper

### How to reproduce

Once slicing is finished (Preview Pane) --> Click Print --> Upload (or) Upload and Print, receive the following error message.  Click okay.  Immediately try again: Print. --> Upload (or) Upload and Print, it works fine.

### Actual results

[image]

### Expected results

Should immediately upload the print file and switch to the Device tab.  Have to re-try uploading before the file is successfully uploaded and the Device tab comes up.

### Project file & Debug log uploads

[debug_Thu_Mar_20_23_24_18_47222.txt](https://github.com/user-attachments/files/19411070/debug_Thu_Mar_20_23_24_18_47222.txt)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

basilfx, 2025-03-31:
I get the exact same error, but retrying does not work.

Downgrading to OrcaSlicer 2.2.0 makes it work again. I am running macOS 15 with the 'allow local network access' flag enabled.

danielskowronski, 2025-05-07:
I have the same issue on v2.3.0 on macOS when connecting to K1C running latest version of Moonraker, but for me it started several months ago. This is not a permanent issue with printer or network as in about 50% of cases it works first time, and in 40% on second attempt (last 10% for few more retries). Orca Slicer is enabled to have local network access in macOS privacy settings and the issue is persistent across multiple macOS versions. Local network is also stable.

[image]

I suspect that the issue may be caused by true connection issue between slicer and printer (after all it's not high-end CPU and WiFi could be better as well), but Orca should allow users to increase internal cURL timeout or offer retry when the issue was on network level (i.e. when we are sure printer didn't get the file). Right now, it's painful to click "OK" on alert window and navigate back to "uplad and print", sometimes several times.

closed 2025-08-14 by github-actions[bot]