OrcaSlicer/OrcaSlicer #9054: All AnkerMake Printer Models no longer display accurate time estimates since using OrcaSlicer 2.3.0 Stable

TheMasterOfTech, 2025-03-24:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0 Stable

### Operating System (OS)

macOS

### OS Version

macOS 15.3.2

### Additional system information

Processor: 2 GHz Quad-Core Intel Core i5
Graphics: Intel Iris Plus Graphics 1536 MB
Memory: 16 GB 3733 MHz LPDDR4X

### Printer

AnkerMake M5 & AnkerMake M5C (affects both models)

### How to reproduce

1. Slice an object, with any amount of instances (whether it be a single instance), etc.
2. Export g-code from Orca
3. Send to printer via AnkerMake Slicer

### Actual results

Time estimates are accurate within slicer, but they no longer are relayed correctly to AnkerMake printers and the printers now show unusual and highly inaccurate time estimate information. This issue was not present in OrcaSlicer 2.2.0 Stable. Attached are example photos of the time estimate information being highly inaccurate on both printers now since using any gcode generated by OrcaSlicer 2.3.0.

### Expected results

AnkerMake printers display accurate time estimate information from any gcode generated by OrcaSlicer like it was in OrcaSlicer 2.2.0 stable.

### Project file & Debug log uploads

[image]

### Checklist of files to include

- [ ] Log file
- [x] Project file

TheMasterOfTech, 2025-03-24:
[ExampleProject.3mf.zip](https://github.com/user-attachments/files/19415188/ExampleProject.3mf.zip)

This is for an M5, but I also have an AnkerMake M5C that is doing now the same behavior since upgrading to 2.3.0 stable.

abaluta, 2025-04-01:
I have the same problem when using a mobile app for printing. Either a huge time or zero. Always appears on different projects and appeared after the update 2.3.0 (printer m5с)

pejuur, 2025-04-18:
I have noticed the same issue. I compared g-code produced with 2.2.0 and 2.3.0 from the same project but did not notice huge differences - only differences were in "EXECUTABLE_BLOCK_START" section where the accelerations values were listed and those were smaller in 2.3.0 version.

In 2.3.0 values are:
machine_max_acceleration_e = 4000
machine_max_acceleration_extruding =   6000
machine_max_acceleration_retracting =   1000
machine_max_acceleration_travel = 6000
machine_max_acceleration_x = 6000
machine_max_acceleration_y = 6000
machine_max_acceleration_z = 300

In 2.2.0 each of these values is 10000

Not tested but I need to test this wheter there is any change in time estimations if I change those acceleration values to match the only value used in 2.2.0.

Also (but not 100% sure) I've noticed that the Orca's own pressure advance test (line) gave the correct time to Ankermake (M5) for some reason. I still have to rerun that test tomorrow to be sure along with the acceleration value test.

pejuur, 2025-05-11:
For a quick solution the .gcode can be manually modified to be more compatible with eufyMake Studio device manager and Ankermake M5 printer (M5 ok, not tested M5C).
In .gcode exported from Orca the estimated time is found in

; estimated printing time (normal mode) = 22m 46s

-tag. And if compared to .gcode created in eufyMake Studio slicer the usage of similar

;TIME:

-tag with the Orca's presentation of time in seconds the Ankermake's Time Remaining display works again. The ;TIME: -tag needs to be added before thumbnail section. For this example 22m 46 is 1366 seconds and if used in .gcode it would be like:
 
; HEADER_BLOCK_END
;TIME:1366s
; THUMBNAIL_BLOCK_START

Also if going further (not fully tested all) some parameter can be to get eufyMake Studio show data more accurately.
If following tag's are added into Orca's .gcode prior ; THUMBNAIL_BLOCK_START -tag:

;Machine Name:AnkerMake M5
;Filament weight: 13g
;speed_infill: 250
;speed_wall_0: 150
;speed_wall_x: 250
;acceleration_print: 1250
;acceleration_travel: 2500
;MAXSPEED:15000
;TIME:1366s

The "Start Printing" and the end-screen of print will show the weight of the filament and print time correctly, also with this example the settings (not sure are all the speed and accelerations needed there) the display in printer show the
x5(250mm/s)
instead of something else. Now the percentage settings for speed work ok while adjusted in printer or in phone app. Usually my speed in Ankermake display was shown:
x1.2(60mm/s)
which in my settings meant the retraction speed, no other speed could match that, and the usage of percentage changes for speed made things quite odd as well where the value was read to the Ankermake display.

Corresponding tags in Orca's .gcode for added tags are:
;Filament weight: -> ; filament used [g]
;speed_infill: -> ; sparse_infill_speed
;speed_wall_0: -> ; outer_wall_speed
;speed_wall_x: -> ; inner_wall_speed
;acceleration_print: -> ; default_acceleration
;acceleration_travel: -> ; travel_acceleration
;MAXSPEED: -> no corresponding value found in Orca's .gcode but should ofcourse be the most fastest speed used in the whole print. I tested this and changed speeds settings in eufyMake Studio slicer but did not find which value really affect to this. But this is the value that defines the speed shown in the Ankermake display but still some "speed" values were needed to be shown in this parameter list in order to speed value is dispalyed.

Do not know wheter there could be any better solutions but I am currently trying to create a post-processing script that adds these tags into .gcode after exporting it from Orca slicer to include all these changes I presented automatically.

zavlaki, 2025-05-23:
I have the same problem with the time stamps. 
[Ankermake M5C with Orca Slicer v2.3.0]

My work around is to add the ;TIME: manually but a bit differently
I have to add this block exactly as it is with the time seconds adjusted to my current project.

; filament_type = "AnkerMake PLA+ Basic"
;TIME:2239s

If the first line with the filament_type is missing somehow it cannot read the following line with the ;TIME:
The ; filament_type = "AnkerMake PLA+ Basic" has to be there.

As other have also found this has to be placed at the start before the ; HEADER_BLOCK_END and the ; THUMBNAIL_BLOCK_START

I hope that works for you too!

revjkramer, 2025-08-09:
The workaround works, but isn't this something OrcaSlicer should have built in for the AnkerMake machines or have they abandoned us like the manufacturer has?