OrcaSlicer/OrcaSlicer #9057: Custome model settings are not loading

luminescence69, 2025-03-24:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0

### Operating System (OS)

Windows

### OS Version

Windows 11

### Printer

Bambu Lab X1C

### How to reproduce

1. Load file from either MakerWorld or 3mf file

### Actual results

The custom settings for the model (layer height, speed, support, etc.) are not loaded into Orca.
When loading the same file into Bambu Studio, the settings are present.

### **From Orca Slicer**
[image]

### **From Bambu Studio**
[image]

### Expected results

Model settings are to be applied automatically upon import

### Project file & Debug log uploads

[debug_Mon_Mar_24_18_31_21_26596.log.0.zip](https://github.com/user-attachments/files/19428075/debug_Mon_Mar_24_18_31_21_26596.log.0.zip)

[Death Whistle.zip](https://github.com/user-attachments/files/19428077/Death.Whistle.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

### Anything else?

This started to happen after the update to 2.3.0

Noisyfox, 2025-03-24:
[image]
Make sure this option is not set to "Load Geometry Only"

Noisyfox, 2025-03-24:
Also the 3mf file you uploaded already have every settings resetted. Could you upload the original file or the download link from MakerWord?

luminescence69, 2025-03-25:
> [image] Make sure this option is not set to "Load Geometry Only"

This fixed the issue.
Thank you very much! 😄

closed 2025-03-25 by luminescence69