OrcaSlicer/OrcaSlicer #9064: Long Slicing times with Organic Supports

irrational314, 2025-03-24:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3

### Operating System (OS)

Linux, Windows

### OS Version

Windows 11 24H2, Ubuntu 24.04.2 LTS

### Additional system information

Laptop:
CPU: Intel i5-1235U
RAM: 16 gb ddr4
GPU: NVidia GeForce RTX 2050 Mobile
Windows 11 24H2, and Ubuntu 24.04.2 LTS

and Desktop:

CPU: AMD Ryzen 7 3700x
RAM: 32GB DDR4
GPU: RX7900xt
Windows 11 24H2 only

### Printer

Cutom IDEX RatRig V-Minion

### How to reproduce

Slice a file with a decent number of Organic supports in 2.2 and 2.3, notice the difference in time to slice, paying attention to how long 2.3 stays on Generating Supports. I have included a project file with the profiles used.

### Actual results

When slicing a file in the latest 2.3 release with organic supports, it takes significantly longer than the 2.2 release with the same file. This only happens with Organic supports. 

Testing of time to slice:
OrcaSlicer 2.2: 3.8 seconds
OrcaSlicer 2.3: 1 minute 48 seconds
PrusaSlicer 2.91: 1.8 seconds

Desktop Tests:
OrcaSlicer 2.2 Release: ~2 seconds
OrcaSlicer 2.3-beta: ~2 seconds
OrcaSlicer 2.3-beta2: ~32 seconds
OrcaSlicer 2.3RC: ~33 seconds
OrcaSlicer 2.3:  ~33 seconds

I included PrusaSlicer just as a reference, for expected performance on a given system, relative to slicer.

### Expected results

Your slicing time for a model with Organic supports should be significantly longer with 2.3 compared to 2.2.

### Project file & Debug log uploads

Project file and OrcaSlicer GCODE exports

[files.zip](https://github.com/user-attachments/files/19431955/files.zip)

### Checklist of files to include

- [ ] Log file
- [x] Project file

### Anything else?

Haven't tested on any other computers. Will work on testing other models.

TheFalldog, 2025-03-24:
Not sure if related, but having somewhat similar issues. 

Slicing step peaks CPU to 100%, then when it gets to optimizing tool path, CPU comes back down then memory usage ramps up to 100% (my system has 32GB) and just sits there for several minutes until I try and cancel and Orca crashes. 

Reverted back to 2.2.0 an am able to slice w/o issue.

Noisyfox, 2025-03-25:
Please try PR https://github.com/SoftFever/OrcaSlicer/pull/9038

irrational314, 2025-04-02:
Tried PR 9038, same slicing times as 2.30. Although some driver changes or maybe just the day for my laptop have increased slicing times five-fold:

PR9038: 5min 17 sec
2.3: 5 min 17 sec
2.2 13.48 sec

However, upon some further testing I think this issue is related to support the vertical tree support from pull #8212, which was brought into the beta in 2.3 beta 2. Combined with that I inadvertently painted on vertical supports:

[image]

Grabbing a different file supports this theory as the slice time without vertical supports is seconds and with a quick but actually not needed tree support added, the slice time balloons significantly:

Test file:
No vertical supports: 4 seconds
Vertical supports: 52 seconds

[test file gcode.zip](https://github.com/user-attachments/files/19570532/test.file.gcode.zip)

I would say for further confirmation; slice a complex file (to naturally increase slice times before vertical supports are added) with and without vertical supports painted on and see the difference in slicing times.

Please me know if I can provide further testing or files.

closed 2025-07-09 by github-actions[bot]