OrcaSlicer/OrcaSlicer #9068: OrcaSlicer 2.2.0 had bug that created best skirt known to humanity

Dr-Wily-Zwei, 2025-03-24:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.2.0

### Operating System (OS)

Windows

### OS Version

windows 11

### Printer

Bambulab P1S

### How to reproduce

Set skirt settings:
Skirt type Per Object
Skirt loops 10
Skirt distance 0.18
Skirt height 20 layers

### Actual results

[image]
OrcaSlicer 2.2.0 would create L-shaped skirt, that uses minimum amount of filament, has the best holding in place properties I've ever tested and is super easy to remove.

### Expected results

I'm printing a lot of tall and slim models and I've tested every single setting that helps with supporting such 3d prints, nothing even comes close to L-shaped Skirt 

### Project file & Debug log uploads

[Cylinder_L_shaped_skirt.zip](https://github.com/user-attachments/files/19439100/Cylinder_L_shaped_skirt.zip)

### Checklist of files to include

- [ ] Log file
- [x] Project file

Noisyfox, 2025-03-25:
Sorry but I don't see any bugs here...?

tlhintoq, 2025-03-25:
Guess:
"had" is past tense.  Maybe 2.3 no longer does this?  The OP doesn't explicitly say that which makes it hard to know what they're reporting.  But if I had to guess, that would be it.

Dr-Wily-Zwei, 2025-03-25:
@Noisyfox @tlhintoq I've updated post to show the same project file sliced in OrcaSlicer 2.2.0 and 2.3.0
What I'm trying to communicate is that "bugged" (L-shaped) skirt that existed in version 2.2.0 was actually way better than "correct" one in 2.3.0.
Once whole bulid plate is filled with slim and tall objects like the one in the first photo, you are saving about 12% of filament and 17% printing time using OrcaSlicer 2.2.0 compared to 2.3.0. Also from my experience that L-shaped skirt provides highest success rate of printing tall and slim objects without tipping them over while printing.

Noisyfox, 2025-03-25:
2.3 added a new option to control this behavior:

[image]

This was added because this single looped version causes unstable skirt for some users.

Dr-Wily-Zwei, 2025-03-25:
@Noisyfox that's it! Case closed