OrcaSlicer/OrcaSlicer #9070: Calibration with PA Pattern freezes OrcaSlicer

Janockel, 2025-03-25:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0

### Operating System (OS)

Windows

### OS Version

Windows 11

### Printer

Bambu P1S

### How to reproduce

Calibration - Pressure Advance - DDE - PA Pattern - OK

### Actual results

After clicking on OK, OrcaSlicer doesn't load the PA Pattern test file an freezes.
Every other PA test works fine.
Since 2.3.0 Official Release the PA Pattern test method freezes OrcaSlicer.
2.3.1 and the nightlys freezes also.
Before 2.3.0 it works also fine.

[image]

### Expected results

The PA Pattern test should generate the calibration file.

### Project file & Debug log uploads

[debug_Tue_Mar_25_07_47_26_37568.log.0.zip](https://github.com/user-attachments/files/19443316/debug_Tue_Mar_25_07_47_26_37568.log.0.zip)

### Checklist of files to include

- [x] Log file
- [ ] Project file

Noisyfox, 2025-03-25:
Could you send me your config folder?

Janockel, 2025-03-25:
The whole folder from %username%\AppData\Roaming\OrcaSlicer?

Noisyfox, 2025-03-25:
> The whole folder from %username%\AppData\Roaming\OrcaSlicer?

Yes

Janockel, 2025-03-25:
The file size is about 45 MB.
Your E-Mail provider says it's too large.

Noisyfox, 2025-03-25:
> The file size is about 45 MB. Your E-Mail provider says it's too large.

Weird... I have a few hundred GB of google cloud storage left, that shouldn't happen. Could you share it via gcloud/mega then?

Noisyfox, 2025-03-25:
Did you just share it via OneDrive?

Janockel, 2025-03-25:
Yes. Does it work?

Noisyfox, 2025-03-25:
> Yes. Does it work?

Yep, I'll take a look soon. Thanks

Noisyfox, 2025-03-25:
Hmmmm odd. Not able to reproduce the crash on my side. Just to confirm, does the selected profiles affect the result on your side? E.g., does changing those values make any differences?

[image]

Janockel, 2025-03-25:
No, changing the profiles doesn't have any effect.
Tested OS 2.3.0-rc and to this version it works without problem.
Copied the pa_pattern.3mf from resources\calib\pressure_advance to the newer versions, but doesn't change the problem.
Don't know why.

Noisyfox, 2025-03-25:
> Tested OS 2.3.0-rc and to this version it works without problem.

2.3rc works fine? That's even more weird...

Janockel, 2025-03-25:
Now i figured out, when i set values to acceleration and speed in the settings and don't let them empty, the pattern is generated and OS don't freeze.
Don't know why i have such a bug.

incur, 2025-03-29:
The same for me, it still worked with the 2.3rc. If I enter the values manually, it does not freeze.

Spike-78, 2025-03-29:
Same fpr me

Noisyfox, 2025-04-02:
So everyone here are having this issue on Windows?

Bersaker, 2025-04-02:
@Noisyfox I have windows 10

incur, 2025-04-02:
Windows 11 for me, yes.

JanB97, 2025-04-04:
yes, same with windows 11

MikeW83, 2025-04-06:
Same here... Windows 11, Orca Slicer 2.3 and PA pattern test freezes orca slicer.

Janockel, 2025-04-07:
> So everyone here are having this issue on Windows?

Windows 11

KaiMildner, 2025-04-09:
Same issue with on 11

AlexanderSalge, 2025-04-13:
In the mean time, which accelerations/speeds should I put there?

Noisyfox, 2025-04-13:
> In the mean time, which accelerations/speeds should I put there?

Whatever accel/speed you want the test to run at.

RcTomcat1, 2025-05-05:
I have the same issue on a windows 11 machine. Leaving speed and acceleration empty freezes the programm

EraZz187, 2025-07-20:
> o acceleration and speed in the sett

Yep solved it

Banabas, 2025-07-29:
Unfortunately, the problem still persists.
I started with a fresh installation.
No changes to the settings.
Habe den Ordner aus APPDATA mit angehängt.

```
Microsoft Windows 11 Pro
Version	10.0.26100 Build 26100
Betriebsystemhersteller	Microsoft Corporation
Systemname	GAMING-PC
Systemhersteller	Micro-Star International Co., Ltd.
Systemmodell	MS-7E12
Systemtyp	x64-basierter PC
System-SKU	To be filled by O.E.M.
Prozessor	AMD Ryzen 7 7800X3D 8-Core Processor, 4201 MHz, 8 Kern(e), 16 logische(r) Prozessor(en)
BIOS-Version/-Datum	American Megatrends International, LLC. 1.J0, 04.07.2025
SMBIOS-Version	3.7
Version des eingebetteten Controllers	255.255
BIOS-Modus	UEFI
BaseBoard-Hersteller	Micro-Star International Co., Ltd.
BaseBoard-Produkt	MAG X670E TOMAHAWK WIFI (MS-7E12)

```

[OrcaSlicer.zip](https://github.com/user-attachments/files/21495587/OrcaSlicer.zip)

Banabas, 2025-10-05:
I'm still having the same problems.
Orca still freezes when I do the PA test.