OrcaSlicer/OrcaSlicer #9081: OrcaSlicer 2.3.0 and previous version crashing upon connecting to printer IP (Linux Mint 22.1)

cobrasvt1999, 2025-03-26:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0

### Operating System (OS)

Linux

### OS Version

Linux Mint 22.1 Cinnamon

### Additional system information

System:
  Kernel: 6.8.0-56-generic arch: x86_64 bits: 64 compiler: gcc v: 13.3.0 clocksource: tsc
  Desktop: Cinnamon v: 6.4.8 tk: GTK v: 3.24.41 wm: Muffin v: 6.4.1 vt: 7 dm: LightDM v: 1.30.0
    Distro: Linux Mint 22.1 Xia base: Ubuntu 24.04 noble
Machine:
  Type: Laptop System: ASUSTeK product: ROG Zephyrus Duo 16 GX650RX_GX650RX v: 1.0
    serial: <superuser required>
  Mobo: ASUSTeK model: GX650RX v: 1.0 serial: <superuser required> uuid: <superuser required>
    UEFI: American Megatrends LLC. v: GX650RX.319 date: 08/11/2023
Battery:
  ID-1: BAT0 charge: 73.7 Wh (100.0%) condition: 73.7/90.0 Wh (81.8%) volts: 15.9 min: 15.9
    model: ASUSTeK ASUS Battery type: Li-ion serial: N/A status: not charging
CPU:
  Info: 8-core model: AMD Ryzen 9 6900HX with Radeon Graphics bits: 64 type: MT MCP smt: enabled
    arch: Zen 3+ rev: 1 cache: L1: 512 KiB L2: 4 MiB L3: 16 MiB
  Speed (MHz): avg: 966 high: 1707 min/max: 400/4935 cores: 1: 400 2: 1618 3: 1397 4: 400 5: 1697
    6: 1707 7: 1397 8: 1669 9: 400 10: 400 11: 1387 12: 400 13: 400 14: 400 15: 1396 16: 400
    bogomips: 105401
  Flags: avx avx2 ht lm nx pae sse sse2 sse3 sse4_1 sse4_2 sse4a ssse3 svm
Graphics:
  Device-1: NVIDIA GA103M [GeForce RTX 3080 Ti Mobile] vendor: ASUSTeK driver: nvidia v: 550.120
    arch: Ampere pcie: speed: 16 GT/s lanes: 8 ports: active: none off: DP-2,eDP-2
    empty: DP-1,HDMI-A-1 bus-ID: 01:00.0 chip-ID: 10de:2420 class-ID: 0300
  Device-2: AMD Rembrandt [Radeon 680M] vendor: ASUSTeK driver: amdgpu v: kernel arch: RDNA-2
    pcie: speed: 16 GT/s lanes: 16 ports: active: DP-3 empty: DP-4, DP-5, DP-6, DP-7, DP-8,
    Writeback-1, eDP-1 bus-ID: 06:00.0 chip-ID: 1002:1681 class-ID: 0300 temp: 54.0 C
  Device-3: IMC Networks USB2.0 HD UVC WebCam driver: uvcvideo type: USB rev: 2.0 speed: 480 Mb/s
    lanes: 1 bus-ID: 5-1:2 chip-ID: 13d3:56eb class-ID: fe01 serial: <filter>
  Display: x11 server: X.Org v: 21.1.11 with: Xwayland v: 23.2.6 driver: X:
    loaded: amdgpu,nouveau unloaded: fbdev,modesetting,vesa failed: nvidia dri: radeonsi
    gpu: amdgpu,nvidia,nvidia-nvswitch display-ID: :0 screens: 1
  Screen-1: 0 s-res: 5083x2700 s-dpi: 191 s-size: 675x372mm (26.57x14.65") s-diag: 771mm (30.34")
  Monitor-1: DP-2 pos: top-right res: 1920x1080 hz: 60 dpi: 142 size: 344x194mm (13.54x7.64")
    diag: 395mm (15.55") modes: N/A
  Monitor-2: DP-4 pos: primary,top-center res: 2560x1600 hz: 60 dpi: 189
    size: 344x215mm (13.54x8.46") diag: 406mm (15.97") modes: N/A
  Monitor-3: DisplayPort-2-2 pos: bottom-l res: 3840x1100 hz: 60 dpi: 287
    size: 340x100mm (13.39x3.94") diag: 354mm (13.95") modes: N/A
  API: EGL v: 1.5 hw: drv: nvidia drv: amd radeonsi platforms: device: 0 drv: nvidia device: 2
    drv: radeonsi device: 3 drv: swrast surfaceless: drv: nvidia x11: drv: nvidia
    inactive: gbm,wayland,device-1
  API: OpenGL v: 4.6.0 compat-v: 4.5 vendor: nvidia mesa v: 550.120 glx-v: 1.4 direct-render: yes
    renderer: NVIDIA GeForce RTX 3080 Ti Laptop GPU/PCIe/SSE2
Audio:
  Device-1: NVIDIA vendor: ASUSTeK driver: snd_hda_intel v: kernel pcie: speed: 16 GT/s lanes: 8
    bus-ID: 01:00.1 chip-ID: 10de:2288 class-ID: 0403
  Device-2: AMD Rembrandt Radeon High Definition Audio driver: snd_hda_intel v: kernel pcie:
    speed: 16 GT/s lanes: 16 bus-ID: 06:00.1 chip-ID: 1002:1640 class-ID: 0403
  Device-3: AMD ACP/ACP3X/ACP6x Audio Coprocessor vendor: ASUSTeK driver: snd_pci_acp6x v: kernel
    pcie: speed: 16 GT/s lanes: 16 bus-ID: 06:00.5 chip-ID: 1022:15e2 class-ID: 0480
  Device-4: AMD Family 17h/19h HD Audio vendor: ASUSTeK driver: snd_hda_intel v: kernel pcie:
    speed: 16 GT/s lanes: 16 bus-ID: 06:00.6 chip-ID: 1022:15e3 class-ID: 0403
[... 94 lines cut ...]

Startup the latest OrcaSlicer appimage, it will boot fine and run fine until you input the local IP:port of your printer. In my case I am using Octo/Kilipper (though the same problem persists with CrealityPrint selected) as the host type and 4408 as the port. OrcaSlicer will lockup immediately and crash within a few seconds.

Hitting the test button will result in the test passing and having no connection issues

### Actual results

After you have input the IP address, Orca slicer will immediately crash upon startup. The only way to actually start OrcaSlicer is to disconnect the computer from the network, start it up so it cannot connect to the printer, then remove the printer IP address. After that, you can again connect the computer to the network and orcaslicer will work properly again. 

### Expected results

OrcaSlicer should continue to run and read information, bed temp, nozzle temp, ect from the printer as well as be able to send STL's to the printer for printing. This does work properly on the windows version per my tests.

### Project file & Debug log uploads

[Error Logs.zip](https://github.com/user-attachments/files/19461677/Error.Logs.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

### Anything else?

If you want any additional data, Please feel free to let me know. If there is anything else I can do from my end to provide you with what you need, give me a shout. Thanks!

Noisyfox, 2025-03-26:
Dup https://github.com/SoftFever/OrcaSlicer/issues/7210

closed 2025-03-26 by Noisyfox

cobrasvt1999, 2025-03-26:
> Dup [#7210](https://github.com/SoftFever/OrcaSlicer/issues/7210)

Damn, Thanks for pointing that one out to me! following that now