OrcaSlicer/OrcaSlicer #9082: Faster PA pattern Calibration

Andreas-strg, 2025-03-26:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

Calibrating Advance PA with PA pattern takes a long time

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

The time and material requirements, as well as readability, would be improved if the Y size of the PA test could be adjusted and reduced using a variable.

[image]

Modern printers no longer require as much speedup way in this case.

buzzhuzz, 2025-04-02:
Pattern width shall be big enough to make head able to reach target speed with target accelerations.

(I just noted that there no any check if head really hits target speed at the tip of the pattern)

Andreas-strg, 2025-04-02:
In this case we could take the acceleration values ​​and calculate the required width (+ little safety)