OrcaSlicer/OrcaSlicer #9085: 0.8mm nozzle causing ineffective flow rate calibration

ohyeah8000, 2025-03-26:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3 stable

### Operating System (OS)

Windows

### OS Version

Win11

### Printer

MKS-Pi CoreXY machine running latest mainline Klipper

### How to reproduce

Flawless operation and calibration possible with 0.4mm nozzle. I selected 0.8mm profile and performed YOLO flow calibration. Then a warning appears and alignment is not possible. Parts are colliding. 
Other users reproduced the issue.

### Actual results

[image]

[image]

[image]

### Expected results

It should look like that no matter what:

[image]

### Project file & Debug log uploads

Please go to the built in Orca calibration folder.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Noisyfox, 2025-03-28:
Could you share your profile? You could just save the project in your first pic as 3mf and upload it here. Thanks.

ohyeah8000, 2025-03-30:
Of course, pardon me.
Here is the project:

[Orca YOLO Flow Calibration_0.8nozzle.zip](https://github.com/user-attachments/files/19524161/Orca.YOLO.Flow.Calibration_0.8nozzle.zip)

I found a workaround by now for this "unexpected feature".
I hit the arrange button and reduce line spacing to value 0 which enables auto line spacing. 
It was a value of 2.19 present which obviouly resulted in an overlap...

Thank you!

Noisyfox, 2025-03-30:
Thanks I'm able to reproduce this problem with your profile. Will take a look soon.

PR #9189 (2025-04-13): Fix flow calibration overlapping objects with nozzle > 0.4mm
Fix #9085

Old:
[image]

New:
[image]
closed 2025-04-13 by Noisyfox