OrcaSlicer/OrcaSlicer #9090: Error with Slicer bed and preview bed not aligning for lulzbot 4 settings in 2.3.0

RubberC0rgi, 2025-03-26:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0

### Operating System (OS)

Windows

### OS Version

Windows 10 19045.5608

### Additional system information

CPU: AMD Ryzen 5 5600X 6-Core 3.70 GHz
Memory: 32 GB 
Graphics Card: PNY RTX 3060TI 
Printer: LulzBot 4

I was trying to use the new Lulzbot settings for OrcaSlicer, and the bed graphic and system are not aligned. 

[image]

No idea about why. 

I asked for help in the discord, and Somone said that the file for the bed was misaligned, so I opened it in Freecad, but it is set to 0,0 by default. So I have no idea what to do. 

Kind regards, 

### Printer

Lulzbot 4

### How to reproduce

1. Select the default Lulzbot 4 and 5 profile
2. See issue. 

### Actual results

this appears with the slicer bed below the actual bed pictured on top. 
[image]

### Expected results

They should align; the printer bed and the slicer bed should be on top of each other. 

### Project file & Debug log uploads

not letting me upload zip file

### Checklist of files to include

- [ ] Log file
- [ ] Project file

wrathernaut, 2025-05-17:
Corrected with new geometry.

[image]

imortal6, 2025-07-22:
how do you fix this miss alignment with taz 4?

wrathernaut, 2025-07-23:
@imortal6 Just download the new bed STL, then go to your bed geometry settings and point to the new STL.

[image]