OrcaSlicer/OrcaSlicer #9092: After reopening project (3mf) buildplates are gone

karambolis8, 2025-03-26:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

since 2.3.0

### Operating System (OS)

Windows

### OS Version

Windows 10 Home

### Printer

Voron Trident 250

### How to reproduce

1. Create multiple print plates in project
2. Arrange models on plates
3. Save project & close Orca
4. Open project in Orca

[image]
[image]

### Actual results

Models are aligned as left, but buildplates are missing

### Expected results

Buildplates are saved along with models in 3mf file.

### Project file & Debug log uploads

[log.zip](https://github.com/user-attachments/files/19474650/log.zip)
[Printing.zip](https://github.com/user-attachments/files/19474649/Printing.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

Noisyfox, 2025-03-27:
[image]
Make sure this option is not set to "Load Geometry Only"

karambolis8, 2025-03-27:
That fixed the problem. I don't remember changing this settings though.

closed 2025-03-27 by karambolis8