OrcaSlicer/OrcaSlicer #9094: 2.3.0 Reproducable memory leak on small 0.8mm slice

Deen-Weible, 2025-03-26:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0

### Operating System (OS)

Linux

### OS Version

Arch Linux 

### Additional system information

Operating System: Arch Linux 
KDE Plasma Version: 6.3.3
KDE Frameworks Version: 6.12.0
Qt Version: 6.8.2
Kernel Version: 6.12.20-1-lts (64-bit)
Graphics Platform: Wayland
Processors: 12 × AMD Ryzen 5 3600 6-Core Processor
Memory: 31.3 GiB of RAM
Graphics Processor: AMD Radeon RX 6750 XT
Manufacturer: Gigabyte Technology Co., Ltd.
Product Name: X570 AORUS ELITE
System Version: -CF

### Printer

Elegoo Neptune 3 Plus

### How to reproduce

Attempt to slice the model (project below)

### Actual results

Gets stuck at 75% slicing and leaks memory until kernel kills OrcaSlicer.

[image]

[image]

### Expected results

It should slice the model and not crash.

### Project file & Debug log uploads

[OrcaSlicer.zip](https://github.com/user-attachments/files/19475645/OrcaSlicer.zip)

(Contains log + project 3mf)

### Checklist of files to include

- [x] Log file
- [x] Project file

Deen-Weible, 2025-03-28:
Now that's interesting. It does work fine with the supports changed to Default (Grid/Organic). 

This is the setup that crashed it (the hybrid trees)
[image]

Noisyfox, 2025-03-28:
> Can reproduce on windows. Only happens on 2.3, not 2.2.
> 
> Something is off with the tree support, 3mf works with organic support style.
> 
> All three tree support optons put` while (conflictQueue.valid()) { LineWithIDs lines = conflictQueue.getCurLines(); float curBottomZ = conflictQueue.getCurrBottomZ(); bottomZs.push_back(curBottomZ); layersLines.push_back(std::move(lines)); }` in ConflictChecker.cpp into infinite loop

Does this fix the issue?
https://github.com/bambulab/BambuStudio/commit/b952006e4db49f00054cc2ac539074222c890d08

Noisyfox, 2025-03-28:
Please give PR #9115 a try

Deen-Weible, 2025-03-28:
Would if I could but docker is giving me problems, Glad it's fixed then. I'll close it up

PR #9115 (2025-04-05): FIX: prevent dead loop of conflict checker
cherry picked from commit bambulab/BambuStudio@b952006e4db49f00054cc2ac539074222c890d08
Thanks BambuLab!

Fix #9094
[image]
closed 2025-03-28 by Deen-Weible