OrcaSlicer/OrcaSlicer #9095: Direction of top/bottom layers

Raaausch, 2025-03-26:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

In Cura there's an option to modify the direction of top and bottom layers. For example I often used 0 45 90 135 degrees. Additionally there's an option to set the direction of the first and last layers (the top and bottom layer that are visible) to a fixed direction so that all layers are pointed in the same direction. It would be nice to have something like this in the future

MxBrnr, 2025-03-27:
You mean like these?

[image]

Raaausch, 2025-04-07:
Yes, but with the added option to modify the angle with which the infill ist rotated. For example 45° instead of the now fixed 90°.

MxBrnr, 2025-04-10:
Ah, yes, that would be a very nice option to have.