OrcaSlicer/OrcaSlicer #9107: Setup Wizard doesn't work

thedizze, 2025-03-27:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0

### Operating System (OS)

Windows

### OS Version

windows 10

### Printer

all printers model

### How to reproduce

If I try open the wizard or i click  Select/remove printer show a black window and nothing happens

[image]

if I try to create printer 

[image]

tried with several brands but the result is always the same 

### Actual results

I'm not able to add printers 

### Expected results

select/remove printer and create printer don't work as expected 

### Project file & Debug log uploads

[log.zip](https://github.com/user-attachments/files/19491415/log.zip)

### Checklist of files to include

- [x] Log file
- [ ] Project file

Noisyfox, 2025-03-28:
Make sure the Orca installation path does not contain any non-ascii characters.

thedizze, 2025-03-28:
path is c:\Program\OrcaSlicer

impala68454, 2025-04-04:
i'm having the same issue 
fresh install of windows 11 
all ms updates installed
visual c++ runtimes installed 
tried orca version 2.3.0, 2.2.0, nightly and portable versions of all three also 
installed in program files and in c:\orca
tried running as admin for both the install and the application 

nothing has worked 

help !

impala68454, 2025-04-06:
> i'm having the same issue fresh install of windows 11 all ms updates installed visual c++ runtimes installed tried orca version 2.3.0, 2.2.0, nightly and portable versions of all three also [...]

so i installed visual studio because i felt like i was missing some library, i selected a number of options so i don't know which one fixed the issue but one of them did  

now its working fine ... apparently there was a dependency that didn't throw an error that i was missing, unfortunately i still don't know what it was 

but everything is now working fine

Noisyfox, 2025-04-06:
Please try installing the latest edgewebview from Microsoft.

impala68454, 2025-04-07:
> Please try installing the latest edgewebview from Microsoft.

tried that ... no help 
it was already on the latest version as i ran all Microsoft updates before i ever tried installing orca 

as i said above it was resolved by installing some part of visual studio

Noisyfox, 2025-04-07:
Which windows version are you using?

impala68454, 2025-04-07:
24H2

Noisyfox, 2025-04-08:
> Win 11 Pro 24H2 26100.3624

Could you zip the entire Orca's log folder and upload it here? Thanks.

impala68454, 2025-04-08:
[log.zip](https://github.com/user-attachments/files/19651204/log.zip)

zipped log file

closed 2025-07-15 by github-actions[bot]