OrcaSlicer/OrcaSlicer #9109: Move Out of Range in project with Prime Tower and Wipe While Retracting

jeffhengesbach, 2025-03-27:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.2.0

### Operating System (OS)

Windows

### OS Version

Windows 11 24H2

### Printer

Voron 2.4 w/ERCF

### How to reproduce

Printing with ERCF (Single Extruder, Multi Material):
Enable both Prime Tower and Extruder Wipe while Retracting
Disabling either option prevents the issue

### Actual results

A negative move / move out of bounds is created as the job transitions from the 1st layer of the prime tower to printing the skirt.  Print job aborts with move out of bounds.

### Expected results

Using Prime Tower with Extruder Wipe while Retracting should not generate a move out of bounds

### Project file & Debug log uploads

[Orca OOB Issue.zip](https://github.com/user-attachments/files/19493438/Orca.OOB.Issue.zip)

### Checklist of files to include

- [ ] Log file
- [x] Project file

Noisyfox, 2025-03-28:
Have you tried 2.3?

Noisyfox, 2025-04-01:
Please give the PR a try

jeffhengesbach, 2025-04-01:
Issue resolved in the PR build

closed 2025-04-13 by Noisyfox

artes44444, 2025-05-14:
I had the same problem but for IDEX printer. This patch fixed the problem. Thanks!

PR #9179 (2025-04-13): Fix wrong travel before printing first skirt point
Fix #9109

don't modify `last_pos`, otherwise it could move out of plate/to wrong places and cause issue

Old:
[image]

New:
[image]