OrcaSlicer/OrcaSlicer #9112: Error when selecting custom filament in Bambu AMS list

conorlmcbride, 2025-03-28:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0 Build 70931e5

### Operating System (OS)

Windows

### OS Version

Windows 11

### Printer

Bambu Lab P1S

### How to reproduce

1. Go to Device Tab
2. Edit an AMS filament
3. Select custom filament (Polymaker CoPE), and choose any color.
4. Press Confirm. The error will appear.

"You need to select the material type and color first."

### Actual results

[image]

### Expected results

Window disappears and new filament is applied, and shown in the AMS list in the device tab.

### Project file & Debug log uploads

Project file: It happens with any project, but here is my current project just in case: [puzzlebox-4m-3N-3H-12b-12c-60h-1B6-0G4-1w2-2t-3z-1M-10X-0p7-0g4-8s-2r-1R666667-e-Z-1T-0y1-0Z2-1.zip](https://github.com/user-attachments/files/19496649/puzzlebox-4m-3N-3H-12b-12c-60h-1B6-0G4-1w2-2t-3z-1M-10X-0p7-0g4-8s-2r-1R666667-e-Z-1T-0y1-0Z2-1.zip)
Log file: [log.zip](https://github.com/user-attachments/files/19496645/log.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Anything else?

I did not see the Polymaker CoPE option while using 2.2.0, so I'm assuming the custom filament showing in the AMS drop down list is new in 2.3.0 as an option, but other "user presets" I have made in the past aren't showing up in the "custom filaments" list. I don't remember doing anything different with the Polymaker CoPE user preset that I didn't do with the others. I will attach a couple screenshots. 

[image]
[image]

Noisyfox, 2025-03-28:
Could you send me your entire configuration folder? Thanks.

conorlmcbride, 2025-03-29:
[OrcaSlicer.zip](https://github.com/user-attachments/files/19514584/OrcaSlicer.zip)
I've only removed the /orcaslicer/plugins/backup directory to keep the .zip small enough to upload to github.

Noisyfox, 2025-03-29:
Able to reproduce the issue, thanks.

Noisyfox, 2025-03-29:
You need to go here and re-add the profile to the printer:
[image]
[image]

Currently, there is a bug that in AMS filament selection it doesn't handle the printer compatibility properly. It does not read the compatible printers from the profile dependencies settings, but only from the profile name. Your profile is named "Polymaker CoPE @Bambu Lab P1S", however it's looking for "@Bambu Lab P1S 0.4 nozzle", which caused this problem. By re-adding the profile, Orca will create a new profile with the full suffix which makes it usable.

closed 2025-04-14 by SoftFever

alerxyan, 2025-04-29:
I'm also experiencing this issue. I tried the nightly build of 2.3.1 and I was now able to select custom filament in the AMS slot, but after a brief second the names and types are replaced with a "?" and the AMS slot greys out (as seen on 4th slot below). Ultimately still unable to use custom filaments in the AMS.

[image]

Maybe a separate but related issue: I have a P1P that has been upgraded to a P1S (I use P1S profiles and have the hardware kit selected on the printer itself), but custom filaments only show up in the AMS selection dropdown when the custom filament profiles dependencies include the P1P. Is that something OrcaSlicer resolves, or is it how the P1P identifies itself to the network plugin?

Noisyfox, 2025-05-08:
> I'm also experiencing this issue. I tried the nightly build of 2.3.1 and I was now able to select custom filament in the AMS slot, but after a brief second the names and types are replaced with a [...]

Which firmware version were you using?

alerxyan, 2025-05-13:
01.07.00.00, haven't yet updated to the newest OTA firmware. Just tried assigning filaments again right before this comment and it appears to be working though. Not sure what's changed...

SoCuul, 2025-12-09:
> Which firmware version were you using?

@Noisyfox I can confirm this happens for me as well. I'm guessing since the upgraded P1S still returns the device identifier of the P1P, and won't accept filament profiles which don't match the device id, even though it acts like a P1S now.

PR #9126 (2025-04-14): Fix AMS filament selection compatible check
Fix #9112