OrcaSlicer/OrcaSlicer #9123: orcalibrate.com Expired

SeeTurtlee, 2025-03-28:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

orcalibrate.com has expired

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

If possible, get the site back up as it is helpful. Possibly start a community fund to help pay for it if funding is the issue. I definitely don't mind chipping in!

Randomblock1, 2025-03-30:
Third party website, not an OrcaSlicer issue. BTW, source here: https://github.com/ItsDeidara/Orca-Slicer-Assistant
Just download and open the HTML.

SeeTurtlee, 2025-03-30:
Awesome! Thank you! i was thinking how it technically isn't orcaslicer but was unsure where else to post it. i figured here was the only place i could think of as the link for the site is in the calibration wiki section. Thank you for the help!!!

closed 2025-03-30 by SeeTurtlee