OrcaSlicer/OrcaSlicer #9127: X-Y hole compensation cause weird line artifact

Pleiades002, 2025-03-29:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0

### Operating System (OS)

Windows

### OS Version

Windows 11

### Printer

Voron 2.4

### How to reproduce

## My STL file
From what I found, if your STL has a hole and X-Y hole compensation is not set to zero value the slicer will create a line artifact on the hole outer wall 
[image]

## X-Y hole compensation = 0.25
[image]

##  X-Y hole compensation = 0
[image]
Reproduce by 
1. Open the attach file
2. Slice the mode with x-y hole compensation set to 0.25 

### Actual results

There is a line artifact appear on the inside of the part  

### Expected results

There shouldn't be any line artifact, and the hole should maintain the correct size through entire heigh

### Project file & Debug log uploads

[X-Y hole compensation artifact.zip](https://github.com/user-attachments/files/19516666/X-Y.hole.compensation.artifact.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

wspock, 2025-06-01:
I had already noticed this problem two versions ago (end of 2024)

ZoskyZosk, 2025-06-01:
I've also noticed artifacts with added material where there shouldn't be any. 

Worse still, in a hole with a small side hole, I have an over-thickness over the entire small hole opening, forming a kind of ring inside, narrowing the diameter.

arronpatton, 2025-08-26:
I agree applying equal and opposite values in the compensation tabs eliminates the anomaly, but it defeats the intended purpose of the offset functionality. Slaving the two values together means I cannot adjust the ID of a hole without simultaneously effecting the dimensions of the part contours. 

I want to print a cube that is 25mm OD with a 10mm hole bore through the middle. Post print dimensions measure a perfect 25mm OD on all external faces but the 10mm bore measures 9.5mm. To resolve this I should be able to simply adjust the x-y hole compensation by +.25mm, but I cannot add in the necessary compensation without also adding in an equal and opposite x-y contour compensation, effectively changing the perfect 25mm OD dimension to a less than perfect 24.5mm OD. 

This is a very simplified example that in practice can be remedied by reaming out the bore to the desired ID, but in many cases, such as the example provided by the OP, reaming a hole is not an option.

Painting on offsets works fine with simple geometries, but with complex geometries this becomes incredibly laborious and again defeats the purpose of having the x-y compensation built into the slicer.

Felix14-v2, 2026-10-30:
Bad bot
2.5.0-dev