OrcaSlicer/OrcaSlicer #9130: Enabling "Only one wall on top surfaces" produces weird inner walls that crash the printer.

ScheglyakStas, 2025-03-29:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0 official release

### Operating System (OS)

Windows

### OS Version

Windows 10 Pro 22H2

### Additional system information

CPU: Ryzen 7 5700G
GPU: integrated
Memory: 16GB DDR4

### Printer

Creality K1

### How to reproduce

Slice the attached STEP file with "Only one wall on top surfaces" enabled.

### Actual results

As you see on the photo, the inner wall on the top surface looks weird, and the printer crashes with "timer too close" error after reaching that wall. 

[image]

### Expected results

The model should be printed without any issues.

### Project file & Debug log uploads

[Magnet cover.zip](https://github.com/user-attachments/files/19521553/Magnet.cover.zip)

### Checklist of files to include

- [ ] Log file
- [x] Project file

### Anything else?

If I disable the "Only one wall on top surfaces" option, the model is printed without issues.

Noisyfox, 2025-04-03:
Seems to be an issue with Arachne wall generator.

Noisyfox, 2025-04-03:
This issue was introduced by commit c7a68b0f26b42cbee5db275ca1adf852b0ab5f93

closed 2025-07-10 by github-actions[bot]