OrcaSlicer/OrcaSlicer #9139: Arachne Causing Constant Z-Hops With Scarf Seam

kcorbett95, 2025-03-29:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0 Build 70931e5

### Operating System (OS)

Windows

### OS Version

Windows 11

### Additional system information

CPU: Ryzen 9 5900x
GPU: RTX 3090
Memory: 64GB DDR4

### Printer

Bambu Lab A1 Mini 

### How to reproduce

Enable Arachne + Scarf Seams  on attached file

### Actual results

Printer will repeatedly slow speed and Z-hop when printing the scarf section, causing blobbing and much slower printing time.

### Expected results

Issue does not occur when Classic wall generation is selected. Scarf sections print at specified speed without stopping and starting and z-hopping constantly.

### Project file & Debug log uploads

[debug_Sat_Mar_29_09_52_12_27904.log.zip](https://github.com/user-attachments/files/19522846/debug_Sat_Mar_29_09_52_12_27904.log.zip)

[Oneup_EDC_Dynaplug_Outer_Cap_w_Taper_-_110mm.zip](https://github.com/user-attachments/files/19522847/Oneup_EDC_Dynaplug_Outer_Cap_w_Taper_-_110mm.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

### Anything else?

See images and videos here:
https://photos.app.goo.gl/SZHb3BeCzKQqBs1T7

Noisyfox, 2025-04-03:
Please give this pr a try https://github.com/SoftFever/OrcaSlicer/pull/9197

kcorbett95, 2025-04-04:
> Please give this pr a try [#9197](https://github.com/SoftFever/OrcaSlicer/pull/9197)

Working! Thanks!

PR #9197 (2025-04-13): Avoid unnecessary travel in scarf seam
 Fix #9139

Old:
[image]

New:
[image]
closed 2025-04-13 by Noisyfox