OrcaSlicer/OrcaSlicer #9144: Crash and termination when calling PA calibration with specific Raise3D machine profile

ohyeah8000, 2025-03-30:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3 stable

### Operating System (OS)

Windows

### OS Version

11

### Additional system information

Thinkpad Intel core I5

### Printer

Raise3D Pro2 with Hyperspeed Update running a Klipper firmware

### How to reproduce

Call PA calibration.
Tower...
Execute.
Program crashes...

[image]

### Actual results

Program shows error message and terminates.

[image]

### Expected results

PA Tower gcode to be created.

### Project file & Debug log uploads

[crash-log.zip](https://github.com

[RaisePro2 Calibration Test.zip](https://github.com/user-attachments/files/19524145/RaisePro2.Calibration.Test.zip)

/user-attachments/files/19524142/crash-log.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Noisyfox, 2025-03-30:
Change the default line width to a non-zero value.

ohyeah8000, 2025-03-30:
Thanks dude!
It works. This is the solution.

[image]

PR #9146 (2025-04-13): Fix PA calib crash when default line width is set to 0
Fix #9144
closed 2025-03-30 by ohyeah8000