OrcaSlicer/OrcaSlicer #9147: CMake 4.0 - Compatibility with CMake < 3.5 has been removed from CMake

lodriguez, 2025-03-30:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

ALL

### Operating System (OS)

Linux

### OS Version

arch linux 6.14

### Printer

ALL

### How to reproduce

1. install cmake 4
2. build the project

### Actual results

Building the deps and the main project fails because cmake_minimum_required is set to something < 3.5

### Expected results

no errors

### Project file & Debug log uploads

CMake Error at CMakeLists.txt:23 (cmake_minimum_required):
  Compatibility with CMake < 3.5 has been removed from CMake.

  Update the VERSION argument <min> value.  Or, use the <min>...<max> syntax
  to tell CMake that the project requires at least <min> but has been updated
  to work with policies introduced by <max> or earlier.

  Or, add -DCMAKE_POLICY_VERSION_MINIMUM=3.5 to try configuring anyway.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Anything else?

to make it build, I had to set -DCMAKE_POLICY_VERSION_MINIMUM=3.5 to deps and orcaslicer cmake command, add it to CMAKE_ARGS in deps/CMakeLists.txt and also patch it in wxWidgets including the cotire modul.

superfloh247, 2025-04-03:
MacOS is also affected

```
./build_release_macos.sh
Build params:
 - ARCH: arm64
 - BUILD_CONFIG: Release
 - BUILD_TARGET: all
 - CMAKE_GENERATOR: Xcode for Slicer, Unix Makefiles for deps
 - OSX_DEPLOYMENT_TARGET: 11.3

Building deps...
+ mkdir -p /Users/lindner/VSCode/OrcaSlicer/deps/build/arm64/OrcaSlicer_dep
+ cd /Users/lindner/VSCode/OrcaSlicer/deps/build/arm64
+ '[' 1. '!=' . ']'
+ cmake /Users/lindner/VSCode/OrcaSlicer/deps -G 'Unix Makefiles' -DDESTDIR=/Users/lindner/VSCode/OrcaSlicer/deps/build/arm64/OrcaSlicer_dep -DOPENSSL_ARCH=darwin64-arm64-cc -DCMAKE_BUILD_TYPE=Release -DCMAKE_OSX_ARCHITECTURES:STRING=arm64 -DCMAKE_OSX_DEPLOYMENT_TARGET=11.3
CMake Error at CMakeLists.txt:23 (cmake_minimum_required):
  Compatibility with CMake < 3.5 has been removed from CMake.

  Update the VERSION argument <min> value.  Or, use the <min>...<max> syntax
  to tell CMake that the project requires at least <min> but has been updated
  to work with policies introduced by <max> or earlier.

  Or, add -DCMAKE_POLICY_VERSION_MINIMUM=3.5 to try configuring anyway.

-- Configuring incomplete, errors occurred!
```

lodriguez, 2025-04-07:
you have to add it to all dependencies and to the dependencies of the dependencies.  You can check out my build script for arch https://git.lod.sh/PKGBUILDs/orca-slicer mainly both the patches and the PKGBUILD.

stephenfeather, 2025-04-11:
For OSX, you can ```export CMAKE_POLICY_VERSION_MINIMUM=3.5``` in the terminal session you are running ```./build_release_macos.sh``` in. This will allow it to cascade down as you are building each of the deps. But honestly, this is the EASIEST bump to overcome.

I spent a lot of time last night fighting with clang and gcc trying get it all to build.

Also, if you are on Xcode 16.x, you may have to bump the target to OSX 14 due to some errors in regard to text transformations.

Pretty much gave up about  30 minutes ago, wasting too much time to patch in a couple of stagnant pull requests and have paying work to do.

~~~
: c++-14 --version
c++-14 (Homebrew GCC 14.2.0_1) 14.2.0
Copyright (C) 2024 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
~~~
~~~
: clang --version
Apple clang version 17.0.0 (clang-1700.0.13.3)
Target: arm64-apple-darwin24.3.0
Thread model: posix
InstalledDir: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin
~~~

closed 2025-05-17 by Noisyfox