OrcaSlicer/OrcaSlicer #9156: 'Fix Model' available to Mac OS users

kunze50, 2025-03-30:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

Lack of cross compatibility

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

Please add to enhancement roadmap:)

stephenfeather, 2025-04-11:
[Model I/O](https://developer.apple.com/documentation/modelio) has been available since OSX 10.11. With that in mind, anyone looking to investigate using this library would have to go back about 10 years to see any 'cool videos' from WWDC or developer examples.  But this is yet another platform-dependent solution.

I'd much rather see the adoption of a cross-platform library that would support Windows/MacOS/Linux. [Meshlab](https://www.meshlab.net/) and [Polygon Mesh Processing Library](https://www.pmp-library.org/) would be viable options.

**BUT** Many a project has gone off the rails by trying to become all things to everyone.

kfbest, 2025-04-26:
Duplicate of https://github.com/SoftFever/OrcaSlicer/issues/8740

kfbest, 2025-04-26:
duplicate of https://github.com/SoftFever/OrcaSlicer/issues/5013 and https://github.com/SoftFever/OrcaSlicer/issues/4923 as well.