OrcaSlicer/OrcaSlicer #9158: Add "Bed type" G-code Placeholder

twintersx, 2025-03-30:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

Currently no existing "bed_type" placeholder in the "Built-in placeholders" advanced G-code tool.

[image]

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

I use multiple different bed surfaces that are different heights. My Klipper print_start macro handles this but it would be nice to pass a predefined "bed_type" placeholder like this:

`BED_TYPE=[current_bed_type]`

and have kliper receive it with something like:

`{% set current_bed_type = params.BED_TYPE|default("") %}
    RESPOND MSG="Current Bed type is: {BED_TYPE}"`

and

` {% if current_bed_type == "Smooth High Temp Plate" %}`

### Describe alternatives you've considered

A working alternative is to change the name of the printer itself for each bed type. Then pass "printer_preset" placeholder.

henrivdr, 2025-03-31:
It's curr_bed_type.
See https://github.com/SoftFever/OrcaSlicer/wiki/bed-types

igiannakas, 2025-04-03:
The below may give you some ideas on how to use the bed type to set your z offsets (also accounting for type of materials).

```
{if (curr_bed_type=="Textured PEI Plate" || curr_bed_type=="Textured Cool Plate")} 
; Gcode offset for Textured plate for all material types
; increasing squish
SET_GCODE_OFFSET Z_ADJUST=-0.045 MOVE=1

; Adjustments for high temp material
; reducing squish
	{if (filament_type[current_extruder]=="ABS" || filament_type[current_extruder]=="ASA")}
	SET_GCODE_OFFSET Z_ADJUST=+0.025 MOVE=1 
	{endif}

	{if (filament_type[current_extruder]=="ABS-GF")}
	SET_GCODE_OFFSET Z_ADJUST=+0.035 MOVE=1 
	{endif}
	
	{if (filament_type[current_extruder]=="PPA-CF")}
	SET_GCODE_OFFSET Z_ADJUST=+0.045 MOVE=1 
	{endif}

	{if (filament_type[current_extruder]=="PET-CF")}
	SET_GCODE_OFFSET Z_ADJUST=+0.050 MOVE=1 
	{endif}

{else} 
; smooth plate selected

; Adjustments for high temp material only
; reducing squish
	{if (filament_type[current_extruder]=="ABS" || filament_type[current_extruder]=="ASA")}
	SET_GCODE_OFFSET Z_ADJUST=+0.035 MOVE=1 
	{endif}

	{if (filament_type[current_extruder]=="ABS-GF")}
	SET_GCODE_OFFSET Z_ADJUST=+0.035 MOVE=1 
	{endif}
	
	{if (filament_type[current_extruder]=="PPA-CF")}
	SET_GCODE_OFFSET Z_ADJUST=+0.035 MOVE=1 
	{endif}

	{if (filament_type[current_extruder]=="PET-CF")}
	SET_GCODE_OFFSET Z_ADJUST=+0.045 MOVE=1 
	{endif}

{endif}
```

yw4z, 2025-04-12:
@igiannakas hi, its a bit off topic but is there a way to get first extrusion point as place holder? i want to move nozzle directly to that point after nozzle clean/wipe end on START_PRINT macro

igiannakas, 2025-04-13:
I don’t think it exists as a placeholder but I’m thinking why do you need it? The first g1 XY Z move is to the first print location immediately after the start gcode completes?

yw4z, 2025-04-13:
i'm currently using first_layer_print_min but its not working good

[image]

as you can see print starts from right but first extrusion point is randomized depends on model

My goal moving nozzle directly to extrusion start point after nozzle wipe. that will remove unnecessary travel and reduce changes leak marks caused by filament leaks from nozzle

[image]

left example current situation. right example how travel should be if i can move nozzle directly to first extrusion point

i have tried to create placeholder before but i was lost in code because conditions for skirt, brim like features

yw4z, 2025-04-13:
btw actual problem for me generated gcode sets Z directly on travel that leads to nozzle scratch on bed
current code on model gcode
```
g1 XY Z
```
should be
```
g1 XY
g1 Z
```

igiannakas, 2025-04-13:
Could you post a project file and a sliced gcode file? Also your print start macro if you don’t mind? I’d like to investigate it in more detail

yw4z, 2025-04-13:
this is a sliced 16cm diameter with height 0.2mm cylinder
g-code file [Cylinder_PLA_18m48s.zip](https://github.com/user-attachments/files/19725549/Cylinder_PLA_18m48s.zip)
project file [Cylinder.zip](https://github.com/user-attachments/files/19725552/Cylinder.zip)

first lines of g-code . its very close to bed for first movement. any small leaks on nozzle leave marks on print bed
```
G1 Z.4 F12000
G1 X47.068 Y47.067
```

printer macros. code based Creality K1 but a lot of modifications by me. POOP macro for nozzle wipe. you can check end of START_PRINT
[print-macros.zip](https://github.com/user-attachments/files/19725522/print-macros.zip)

this is machine start gcode
```
M140 S0
M104 S0
START_PRINT EXT_TEMP=[nozzle_temperature_initial_layer] BED_TEMP=[bed_temperature_initial_layer_single] AREA_MIN={first_layer_print_min[0]},{first_layer_print_min[1]} AREA_MAX={first_layer_print_max[0]},{first_layer_print_max[1]}
RESPOND PREFIX=// MSG="Starting Print..."
```

here is schematic from side view
[image]

yw4z, 2025-04-13:
here is more explanatory one but this is one of problem
[image]

DoctorWiz, 2025-05-12:
> It's curr_bed_type. See https://github.com/SoftFever/OrcaSlicer/wiki/bed-types

When I attempt to use this in Machine start G-code I get the error "Failed to generate G-code for invalid custom G-code."
Also, it doesn't show up anywhere in the 'Built-in placeholders' list.  Nor is anything else that would seem to relate to the bed type.

This would be handy to have, as I typically adjust my Z-Offset slightly for smooth vs. textured build plates.