OrcaSlicer/OrcaSlicer #9162: Adjusting Seam Gap Creates Big Holes in Walls

darkspartan239, 2025-03-31:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.2.2

### Operating System (OS)

Windows

### OS Version

Windows 10

### Printer

Qidi Plus 4

### How to reproduce

1. Go to Quality
2. Scroll down to seam
3. adjust seam gap
4. slice file

Slicing immediately without changing the seam gap produces a normal gap but I need to adjust the gap for a 0.6mm nozzle

### Actual results

After adjusting the seam gap and slicing the file it results in a huge hole on the inner and outer walls where the gap should be, reference the screenshot below

[image]

### Expected results

After adjusting seam gap to 5% the gap should shrink from the default of 15% instead of creating a huge gap.

### Project file & Debug log uploads

[Seam Gap Issue Project.zip](https://github.com/user-attachments/files/19529347/Seam.Gap.Issue.Project.zip)

[debug_Sun_Mar_30_21_30_23_10404.log.zip](https://github.com/user-attachments/files/19529371/debug_Sun_Mar_30_21_30_23_10404.log.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

darkspartan239, 2025-03-31:
I accidentally uploaded an image of an input shaper graph.  Here is the image of the issue 

[image]

discip, 2025-03-31:
@darkspartan239
If you want it be **5%** you need to enter **`%`** after the value, otherwise it will be interpreted as **`mm`**. 😊

discip, 2025-03-31:
@darkspartan239
If you want it be **5%**, you need to enter **`%`** after the value, otherwise it will be interpreted as **`mm`**. 😊

igiannakas, 2025-03-31:
> [@darkspartan239](https://github.com/darkspartan239) If you want it be **5%**, you need to enter **`%`** after the value, otherwise it will be interpreted as **`mm`**. 😊

This! Without the % it is in mm.

darkspartan239, 2025-03-31:
> @darkspartan239
> If you want it be **5%**, you need to enter **`%`** after the value, otherwise it will be interpreted as **`mm`**. 😊

Wow I feel dumb, sorry to trouble you guys with that. Thank you for the help.

closed 2025-03-31 by darkspartan239