OrcaSlicer/OrcaSlicer #9163: [Flatpak] Preview window issue with expanding

Schildkroet, 2025-03-31:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.1-dev

### Operating System (OS)

Linux

### OS Version

Ubuntu 24.04

### Additional system information

CPU: AMD Ryzen 7 5800X (16) @ 4.85 GHz
GPU: AMD Radeon RX 6650 XT [Discrete]
DE: GNOME 46.0
WM: Mutter (Wayland)

### Printer

Bambu Lab A1

### How to reproduce

Slice any model

### Actual results

In preview window, the information overlay (e.g. line typ, flowrate etc) is constantly expanding to the left as long as you move the mouse. 

### Expected results

The information overlay does not expand.

### Project file & Debug log uploads

https://github.com/user-attachments/assets/04a25fb6-7e3a-48c8-83b5-0bfa2ad3d07f

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Noisyfox, 2025-04-02:
Please try PR #9171

Schildkroet, 2025-04-02:
PR solves the problem for me

closed 2025-04-10 by Noisyfox