OrcaSlicer/OrcaSlicer #9169: Crash when opening the "add/remove filament" menu

splatnt, 2025-03-31:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0

### Operating System (OS)

Windows

### OS Version

Windows 11 24H2

### Additional system information

Ryzen 9 6900HS
Radeon RX6700S
16gb DDR5

### Printer

Bambu A1 Mini

### How to reproduce

1. Go to the prepare screen
2. Click on the filament dropdown
3. Click the "Add/Remove filament" option

### Actual results

Program freezes and crashes

### Expected results

Program should open filament menu

### Project file & Debug log uploads

[log.zip](https://github.com/user-attachments/files/19540715/log.zip)

### Checklist of files to include

- [x] Log file
- [ ] Project file

### Anything else?

Issue still happening on fresh install

jordosaxman, 2025-04-07:
Opening menus like Filament (set filaments to use) and Printer settings results in a LONG pause (~30 seconds) for me, that would make me think it had crashed.  I'm using an HP Intel x64 Windows machine.  You press the button and it does nothing for a long time.  If you click anywhere in Orca until it opens the menu, it looks like it's crashed (menu bar changes colors, goes unresponsive).  They will eventually open but the delay is excessive.  Once you've opened either menu once, it takes less time the next time (~3 seconds) until Orca is restarted, then the excessive delay is re-introduced.

closed 2025-07-14 by github-actions[bot]

kraftaksvk, 2025-09-02:
can confirm this is still happening.