OrcaSlicer/OrcaSlicer #9172: Travel still calculated for timelapse even in "traditional" mode

Jorvalt, 2025-04-01:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0

### Operating System (OS)

Windows

### OS Version

Windows 10

### Printer

Bambu Lab A1

### How to reproduce

1. Slice any print
2. Check "travel" under line type

### Actual results

The slicer thinks the print head will move to the purge chute regardless of timelapse setting, as evidenced by the travel going all the way to the left after every layer.

[image]

### Expected results

If "traditional" is selected, this travel should not happen because the printhead stays where it is when it takes a picture for the timelapse.

### Project file & Debug log uploads

[Cube.zip](https://github.com/user-attachments/files/19543295/Cube.zip)

### Checklist of files to include

- [ ] Log file
- [x] Project file

### Anything else?

This isn't a major issue as it doesn't affect the print itself, but it throws off the time estimate and on larger objects or lower layer height prints it can be very far off when it comes to estimated time for the print.

Jorvalt, 2025-04-01:
I don't know for sure but this may also throw off the arrange function as it expects so much travel from the left side of the plate.

Noisyfox, 2025-04-04:
That's normal behavior for A1, see https://wiki.bambulab.com/en/software/bambu-studio/Timelapse

closed 2025-04-04 by Noisyfox