OrcaSlicer/OrcaSlicer #9178: Seam gaps (or wall overlaps) on bottom layers in vase mode

crooteam, 2025-04-01:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0

### Operating System (OS)

Windows

### OS Version

11

### Printer

Flyingbear S1, creality ender 3 v3 se (on klipper)

### How to reproduce

perimeters of walls do not connect. seam gap setting does not work in vase mode

### Actual results

perimeters of walls do not connect. 

### Expected results

perimeters of walls do not connect. seam gap setting does not work in vase mode

### Project file & Debug log uploads

perimeters of walls - connect

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Noisyfox, 2025-04-01:
Are you able to share a 3mf file of your project? Thanks

crooteam, 2025-04-16:
> Are you able to share a 3mf file of your project? Thanks

https://disk.yandex.ru/d/W6NAEo-WMP5VZQ

DangerManiac, 2025-04-21:
Is seam gap working at all? Did a test from 0 to 16% and there was no difference at all in the seam.

closed 2025-07-28 by github-actions[bot]