OrcaSlicer/OrcaSlicer #9182: Mouse curser covers tooltips

briped, 2025-04-01:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0 Build 70931e5

### Operating System (OS)

Windows

### OS Version

Windows 11

### Additional system information

Although I don't think it's relevant for the issue, here it is:
| Item | Value |
| ---: | :--- |
| CPU | AMD Ryzen 7 3700X |
| RAM | 64G |
| GPU | nVidia GeForce RTX 2070 Super |

### Printer

Creality K1 Max

### How to reproduce

1. Go to the Prepare tab and add/select any object on the print plate.
2. Hover the mouse cursor over a tool in the toolbar.

### Actual results

Tooltip is covered by the mouse cursor. Larger cursor covers more of the tooltip.

### Expected results

Tooltips should not be covered by the mouse cursor. It would be better to have them slightly below the "butt" or slightly above the "point" of the cursor.

### Project file & Debug log uploads

Not related to a specific project, and not really an error, more of a design oversight,

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Anything else?

I have not tested it, but of things I imagine could have an impact is that I have scaling set to 200% and increased the size of the cursor, otherwise things are too small for me on my 4K monitors.

I've attached a screenshot of one of the more annoying places I've encountered this. However, the only tool (IrfanView) I readily found that would also capture the pointer, apparently captures a regular sized mouse pointer- Mine is double the size shown in the screenshot.

[image]

closed 2025-07-09 by github-actions[bot]

kazafog, 2025-08-10:
how do we reopen it? Bambu labs also does not seem to think this is important. However people like myself who are getting on and require a large mouse pointer will struggle to learn to use this great software purely because the tooltips are hidden. You may as well remove the tooltips because they are useless to me at the moment. sure offsetting the position would be simple, or push them to the forefront above the pointer.

NilsRo, 2025-09-28:
In my setup sometimes nothing is shown with very high resolution. Would be nice to get it solved and should be easy perhaps to move every tooltip with an offset to the right...

[image]