OrcaSlicer/OrcaSlicer #9183: Bridges sliced with Inner wall width

dzid26, 2025-04-01:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.3.0

### Operating System (OS)

Windows

### OS Version

Win11

### Printer

SV07+

### How to reproduce

1. Load Autodesk FDM test handy model
2. Set Inner wall to 0.7mm

### Actual results

Line width
[image]

### Expected results

External wall bridges shouldn't be thicker.

### Project file & Debug log uploads

Autodesk FDM test handy model

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Anything else?

Line type:
[image]

Settings:

[image]

igiannakas, 2025-04-02:
That’s not a bridge that is an overhang wall.

dzid26, 2025-04-02:
> That’s not a bridge that is an overhang wall.

Is my screenshot confusing due to camera tilt?  Or do you disagree with https://github.com/SoftFever/OrcaSlicer/issues/2231 issue ? 

It's a straight bridge - mislabeled by Orca as the overhang.

igiannakas, 2025-04-02:
A bridge has two parts - the perimeters and the solid infill. The solid infill is what becomes a bridge. The perimeters are always classified as overhang walls. The overhang walls can be printed with bridge parameters is disabling curled overhang detection or as overhang walls if enabled.

The reason for this is to preserve wall extrusion spacing, and width uniformity between walls and overhang walls.

dzid26, 2025-04-03:
I understand what you are saying but I think the Line Type display should be displayed as _Bridge_ when Bridge parameters are used to print it (with curled overhang detection off - as you say). I will leave that to be discussed in https://github.com/SoftFever/OrcaSlicer/issues/2231#issuecomment-2770801358 though. 

Regardless, the external perimeter of the overhang should not change to the thickness of Inner wall width:
[image]
> width uniformity between walls and overhang walls 

   --    is not maintained.

igiannakas, 2025-04-03:
Try with thick bridges turned off

dzid26, 2025-04-03:
Thick bridges were already off. 

Turns out this issue doesn't occur with Arachne wall generator.

closed 2025-07-10 by github-actions[bot]