OrcaSlicer/OrcaSlicer #9184: Fundamental speed capping (& flow rate) calculation in gcode.cpp is wrong, high flow printing issues

ShantanuNair, 2025-04-01:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

Release 2.3.0

### Operating System (OS)

macOS

### OS Version

MacOs Sonoma

### Printer

Ender3V2 (Voxelab Aquila X2)

### How to reproduce

Set max volumetric speed on a print (a voron cube with 1 wall should work for 0.6mm nozzle and all line widths 0.63) to assumed hotends max for given filament and temp.
Set high flow ratio above 1, something like 1.1
Set all printing speeds to near max volumetric speed / (line width * layer height)

Watch in preview how there are several features printed at flows above max set. Much above. The max changes proportional to flow ratio. So to get expected max volumetric speed we have to divide it by flow ratio, and voila, all flows are calculated properly, because now speed is capped properly (explained), and we don't get flow rates adhering to our max volumetric speed because speed capping works as expected.

### Actual results

Flow rates are not clamped by filament_max_volumetric_flow properly.

### Expected results

Correct flow rates achieved and correct speed capping for much better print quality when issues are caused by printing at high flow and near the hotends flow limit for that filament.

### Project file & Debug log uploads

[bad-flows-demo.3mf.zip](https://github.com/user-attachments/files/19558462/bad-flows-demo.3mf.zip)

[debug_Tue_Apr_01_19_44_24_87340.log.0.zip](https://github.com/user-attachments/files/19558473/debug_Tue_Apr_01_19_44_24_87340.log.0.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

### Anything else?

Continuing off from [here](https://github.com/SoftFever/OrcaSlicer/issues/7307#issuecomment-2770515402), as I've isolated the issue.

The speed cap is implemented incorrectly as per my understanding of how it should work. Whether arachne or classic is chosen wouldn't affect as it's a core calculation for capping flow that's done wrong. The current bug lets the effective flow exceed the user‑specified max by a factor equal to the flow ratio.

Now, let's go through gcode.cpp, some of what looks like to be code common to prusa/orca/bambu,  regarding max volumetric speed:
https://github.com/SoftFever/OrcaSlicer/blob/70931e5321fa66966a5bfb251efca0e82307d427/src/libslic3r/GCode.cpp#L5215
Here, _if_ I understand correctly, we can see that flow ratio is applied immediately.
https://github.com/SoftFever/OrcaSlicer/blob/70931e5321fa66966a5bfb251efca0e82307d427/src/libslic3r/GCode.cpp#L5318-L5324
Later, we enforce the cap like so.

This is where I think the bug is, I'll explain below, and it's present all the way through Prusa, Orca and Bambu, so I can understand why it's hard to believe it may be a flow calculation issue too, I was unsure myself, but following the code, it looks like a bug.

### Explanation:

Look at the code enforcing the cap - **the max volumetric speed is compared against a value that has already been multiplied by the flow ratio.** 

Assume the following:
`Max Volumetric Speed: 25 mm³/s`
`Flow Ratio: 1.125`
`Base Flow: Let this be B (in mm³/mm)`

If `path.mm3_per_mm` is the base flow and you set a flow ratio of 1.125, then `_mm3_per_mm` is 1.125 * base. This means the allowed travel speed is reduced by the multiplier. However, when you then multiply the capped speed by the effective (already scaled) flow, the resulting effective volumetric flow ends up higher than intended.

$$\text{Effective Flow} = B \times 1.125$$
$$\text{Correct Allowed Speed Calculation} = \frac{25}{B \times 1.125}$$

However, in gcode.cpp, only the base flow (ignoring the flow ratio) is used to enforce the cap, and so the (incorrect) allowed speed is computed as:

   $$\text{Incorrect Allowed Speed Calculation} = \frac{25}{B}$$  

This value is higher by a factor of 1.125 compared to the correct allowed speed.

When you then multiply the incorrect allowed speed by the effective flow, the actual volumetric flow becomes:
Actual Volumetric Flow =    $$\left(\frac{25}{B}\right) \times (B \times 1.125) = 25 \times 1.125 = 28.125\ \text{mm}^3/s$$  

Which is higher than the intended 25 mm³/s.

@SoftFever Please do take a look, or let me know who to tag if someone else should look at it, I think I'm onto something here. Let me know how I can help, and if it's not clear if it's a bug, apologies, I may be wrong. 

**If it's indeed an issue, this will also fix issues downstream in arachne/classic generation since they would use the same extrusion flow values, fixing fundamental flow calculations. Fixing it will lower the allowed travel speeds (and thus extrusion rates) in Arachne’s generated paths, ensuring consistency, similarly for Classic as well. For those printing near the upper limit of their hotend's flow rate, the fix would reduce the allowed travel speeds to allow so that flows don't exceed the max flow, allowing for much better high flow/speed printing!**

**My proposed fix would be simple:
Divide by Effective flow rate when capping speed, as in the correct expresion above.**

I will create a PR soon, and will need help testing if this is confirmed bug.

Apologies for double posting this in the that issue, I decided to make a standalone issue given that thread has become noisy, and people are conflating this with the UI not accurately displaying flow in preview.

ShantanuNair, 2025-04-03:
@SoftFever @Noisyfox I actually did dig in deeper and think I may have found some bugs in core gcode.cpp which cause speed capping to be enforced incorrectly. The effect is that since line widths are already calculated, and the speed cap sets the speed lower than it should, the flow will be higher.

I am digging into the details of this, and figuring out just how to consolidate changes, and will need help later figuring out downstream effects, but I wanted to ask a critical question:

Do you intend to treat filament_flow_ratio and print_flow_ratio the same? Are you considering filament_flow_ratio to be only a calibration step for extruder feedrate, and print_flow_ratio to be an intentional overflow from the user (for dimensional acc etc)?

I'm trying to wrap my head around implications and it's tricky. Question is, if we consider filament_flow_ratio to be just a calibration step, and it seems to me - we do, in core places, but not in some others (I'm unsure).

Now, if we treat filament flow ratio as “upstream calibration” the slicer should slicer illusion that we have the correct spool diameter or setting for extruder skipping. That way, path.mm3_per_mm is already correct physically. By the time we do speed capping, _mm3_per_mm includes the effect.

From what I understand, no one uses it like so, but this is what Orca's implementation assumes about how it's used: 
Let’s say your slicer geometry says “we need 2mm³/s.” If you
	•	Have filament_flow_ratio = 1.10 because your extruder steps are off by 10%. That means the slicer’s G‐code will say “we want 2mm³/s physically, but we have to command 1.10× the steps to get that 2mm³/s.” In that scenario, the slicer does not want to reduce speed, because physically your nozzle is still pushing 2mm³/s.
	•	Have print_flow_ratio = 1.10, that means “please extrude 10% more plastic than the geometry’s baseline. So physically you’re at 2.2mm³/s.” Now the slicer sees this as an actual higher volume and might do a volumetric speed cap to avoid exceeding 15mm³/s or whatever the user set.

Let me know if I'm gravely misunderstanding things.

In reality filament_flow_ratio is the only one set by users, and if they set print_flow_ratio, it is under the impression it does the same thing. 

They are both different. People are using filament_flow_ratio, instead of print_flow_ratio in order to intentionally extrude more plastic (not an upstream calibration), and so are hitting flow rates extruding more plastic than intended because the slicer is setting too high a speed thinking that the filament ratio is only set with the aim to get the right flow, not realizing that its meant to limit speeds using that ratio.

Am I making sense? I don't work in cpp often, and only just got into 3d printing a couple weeks ago when I got a free ender 3v2, and just got klipper installed :) So do bear with me, I know this is a rambling wall of text, but I am trying to see if there is a way to consolidate this so it works as expected for users.

Noisyfox, 2025-04-04:
Technically speaking you are correct about how these two settings are designed for different purposes. Though as you also pointed out, (at least some of the) people might not realise that and causes confusion. Especially with the fact that the `print_flow_ratio` is only available when you switch to per-object settings, so I gaurantee you that a lot of users might even never know this option even exist.

Another consideration is, is this behavior (deliberately extrude more plastic) encouraged? Is this a normal/common practice in real world scenario? If so, we might want to tweak the UI so it's more accessable and less confusion.

dewi-ny-je, 2025-04-04:
Deliberate over extrusion is used mostly for watertight prints, judging from my experience and online videos (and I watched many) it's uncommon.

But the gcode preview showing the wrong flow is affecting everyone.

dewi-ny-je, 2025-04-09:
See my answer there.

Capping should take place before filament_flow_multiplier, just like the preview should (which would have avoided your troubles altogether).
For SURE the filament_flow_multiplier should NOT be used to cap the gcode.

print_flow_multiplier is a different thing which can be equally applied pre- or post-capping, depending how we interpret it.

Is it a tweak/testing parameter? then it must allow the user to do whatever they want.
Is it something used by the users to make some parts "stronger" or "weaker"? then it should obey the max_flow_rate capping, which is the max the printer can deliver to get good prints, so for those "stronger" parts with a modifier >1 the speed should be reduced to obey the max flow rate the printer can reliably provide.

ShantanuNair, 2025-04-14:
> Another consideration is, is this behavior (deliberately extrude more plastic) encouraged? Is this a normal/common practice in real world scenario? If so, we might want to tweak the UI so it's more accessable and less confusion.

@Noisyfox Absolutely, 100% imo. 

See how everyone's using filament flow ratio, when it doesn't get accounted for by speedcap. Most >1 values, will lead to incorrect extrusions wrt user expectations. If a user sets filament flow ratio to something other than 1, and then does a YOLO test, their results are WRONG, since speedcap takes into account only print flow ratio, but extruder steps uses both ratios. The user thinks setting the resultant yolo flow ratio at filament flow ratio is equivalent to wha the test does, but it's all off imo. The print ratio tooltip says both ratios are just multiplied, causing this confusion. ALL youtube max flow guides / flow rate / extrusion multiplier guides ask users to set the filament flow ratio. So do Orca docs. Users should set print_flow_Ratio instead, per object, if they want to replicate flow calib test extrusion profiles. Users are surprised when volumetric max speed isn't enforced after setting filament flow ratio. A fil flow ratio >1 WILL cause overextrusion (too high speed relative to e_per_mm) whenever extruder isn't skipping. If <1, and the extruder isn't overfeeding, the speed will be capped too low, since it thinks there's less plastic flowing through the nozzle than there is. The calibration tests v high >1 values up to 1.20. These WILL change plastic flow at the nozzle since there are more esteps per mm, but speed is capped as if it's only geometric volume flow. Almost everyone uses orca calibration following a guide, and without understanding what/how design considerations have been ported from slic3r. All those guides calibrate using tests where speedcap is enforced with the ratio, and then update configs such that actual printing will extrude differently because of speedcap implementation.

ShantanuNair, 2025-04-14:
Closing similar issues due to noise. All related discussions regarding this continued here https://github.com/SoftFever/OrcaSlicer/pull/9218. And flow calibration related impact of different flow ratios, and whether fix applies is followed here: https://github.com/SoftFever/OrcaSlicer/issues/9261

closed 2025-04-14 by ShantanuNair