OrcaSlicer/OrcaSlicer #9194: A1 Mini reproducably misses grate during cleaning

Pandapip1, 2025-04-02:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0

### Operating System (OS)

Linux

### OS Version

Linux gavin-laptop-nixos-1 6.14.0 #1-NixOS SMP PREEMPT_DYNAMIC Mon Mar 24 14:02:41 UTC 2025 x86_64 GNU/Linux

### Printer

A1 Mini. Possibly also X1C; will update with results.

### How to reproduce

1. Home the printer manually using the menu (optional)
2. Print like usual
3. Observe the cleaning process

### Actual results

The printer misses the grate and pushes and scrapes the nozzle into the print bed. See attached images.

[image]
[image]

### Expected results

The printer should scrape the nozzle against the grate.

### Project file & Debug log uploads

[lanyardmount.scad.3mf.PTcYGi.zip](https://github.com/user-attachments/files/19575896/lanyardmount.scad.3mf.PTcYGi.zip)

```
~ 
❯ cat $HOME/.config/OrcaSlicer/log/debug_Wed_Apr_02_13_46_41_451776.log.0 
[warning]       2025-04-02 13:46:42.502691[Thread 0x00007f311cdea980]:get_network_function, can not find function bambu_network_del_subscribe
[warning]       2025-04-02 13:46:42.503105[Thread 0x00007f311cdea980]:get_network_function, can not find function bambu_network_get_model_id_from_desgin_id
[warning]       2025-04-02 13:46:42.503176[Thread 0x00007f311cdea980]:get_network_function, can not find function bambu_network_get_profile_3mf
[error] 2025-04-02 13:46:45.720903[Thread 0x00007f311cdea980]:Unable to open the file/run/media/gavin/USB DISK/pdu-prototype-0.3mf
[warning]       2025-04-02 13:46:46.232172[Thread 0x00007f311cdea980]:13:46:46: Navigation request to 'file:///nix/store/lmjvznp3qmknd8sd1ay82nimcgpai1yc-orca-slicer-v2.3.0/share/OrcaSlicer/web/homepage/index.html?lang=en_US' (target='')
[warning]       2025-04-02 13:46:46.270244[Thread 0x00007f311cdea980]:13:46:46: Navigation complete; url='file:///nix/store/lmjvznp3qmknd8sd1ay82nimcgpai1yc-orca-slicer-v2.3.0/share/OrcaSlicer/web/homepage/index.html?lang=en_US'
[warning]       2025-04-02 13:46:46.452471[Thread 0x00007f311cdea980]:13:46:46: Script message received; value = {"sequence_id":1743626806,"command":"get_login_info"}, handler = 
[warning]       2025-04-02 13:46:46.452543[Thread 0x00007f311cdea980]:13:46:46: Script message received; value = {"sequence_id":1743626806,"command":"get_recent_projects"}, handler = 
[warning]       2025-04-02 13:46:46.453524[Thread 0x00007f311cdea980]:13:46:46: Script message received; value = {"sequence_id":1743626806,"command":"modelmall_model_advise_get"}, handler = 
[warning]       2025-04-02 13:46:46.453608[Thread 0x00007f311cdea980]:13:46:46: Document loaded; url='file:///nix/store/lmjvznp3qmknd8sd1ay82nimcgpai1yc-orca-slicer-v2.3.0/share/OrcaSlicer/web/homepage/index.html?lang=en_US'
[warning]       2025-04-02 13:46:49.484934[Thread 0x00007f1ffe7fc6c0]:[Orca Updater]: online version=1.10.1.1, current_version=1.10.1.1, no need to download
[warning]       2025-04-02 13:46:52.537845[Thread 0x00007f311cdea980]:13:46:52: Script message received; value = {"sequence_id":1743626813,"command":"homepage_openproject"}, handler = 
[warning]       2025-04-02 13:47:30.915958[Thread 0x00007f311cdea980]:13:47:30: Script message received; value = {"sequence_id":1743626851,"command":"homepage_newproject"}, handler = 
[error] 2025-04-02 13:47:47.859215[Thread 0x00007f311cdea980]:File Metadata/plate_1.png not found from archive
[error] 2025-04-02 13:59:10.833853[Thread 0x00007f311cdea980]:calc_exclude_triangles:Unable to create exclude triangles
```

### Checklist of files to include

- [x] Log file
- [x] Project file

johnny-bit, 2025-04-03:
Looks like it's off by 3mm.

Pandapip1, 2025-04-07:
Looks like the X1 Carbon also smashes the nozzle into the bed, although I'm not sure it has a grate to clean the nozzle in the first place. Not sure if this is intended behaviour. 

[image]

[image]

johnny-bit, 2025-04-07:
> Looks like the X1 Carbon also smashes the nozzle into the bed,

No, that's the mark of plate detection. Since I post stuff to makerworld I have dedicated printer for that and use BS on that and I can tell you that i have the build plate wearing out in the same spot :) It's even worse on A1 since A1 uses that to clean the nozzle :)

Pandapip1, 2025-04-09:
Update: observed another builtin print on the A1 mini. As @johnny-bit noted, t seems that it's off by about 3mm. However, it's off by about 3mm in both the X and Y axes.

Pandapip1, 2025-07-09:
Haven't tried it in a while, and my A1 is over 1000 mi away so I can't really test. Pretty sure this would have already been closed if the bug had been fixed tho.

Pandapip1, 2026-10-30:
Unless someone's fixed this, this is still an issue