OrcaSlicer/OrcaSlicer #9198: Excluded bed area distortion angles

living-relation, 2025-04-03:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0

### Operating System (OS)

Windows

### OS Version

Windows 11

### Additional system information

Lenovo Thinkpad
Intel(R) Core(TM) i7-8565U CPU @ 1.80GHz   1.99 GHz
8.00 GB RAM
Intel UHD Graphics 620
Windows 11 Home 24H2

### Printer

Creality K1

### How to reproduce

Rectangle Bed printable area 225 x 204 (doesn't affect the problem no matter what dims you use).
Texture and Model both "none" (although this doesn't affect it either).

[image]

[image]

[image]

### Actual results

Why is it not showing a rectangle for the Excluded Bed Area?

### Expected results

it should be rectangular

### Project file & Debug log uploads

not needed

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Noisyfox, 2025-04-03:
Please give this a try https://github.com/SoftFever/OrcaSlicer/pull/8792

living-relation, 2025-04-03:
> Please give this a try [#8792](https://github.com/SoftFever/OrcaSlicer/pull/8792)

I am using simple rectangle already. And have no texture or bed model loaded (although the odd shape was still showing when I did have bed texture and model loaded)

living-relation, 2025-04-03:
Also, regardless of the overall size of my excluded bed area, the twist and corner distortions remain unchanged. 

Why is it twisted like that and what would cause the spurs at the corners too?

Noisyfox, 2025-04-03:
> Also, regardless of the overall size of my excluded bed area, the twist and corner distortions remain unchanged.
> 
> Why is it twisted like that and what would cause the spurs at the corners too?

Because that was designed by BambuLab for their printers only.

living-relation, 2025-04-03:
> > Also, regardless of the overall size of my excluded bed area, the twist and corner distortions remain unchanged.
> > Why is it twisted like that and what would cause the spurs at the corners too?
> 
> Because that was designed by BambuLab for their printers only.

Can you explain why exactly? I don't see how that could affect the display of a rectangle being projected onto another rectangle. Everything is based on xy coordinates (or appears to be) wrt the origin at 0,0.

living-relation, 2025-04-03:
> The code was designed so that when you enter 4 points in counter clockwise, it add
> 
> So the first point MUST be at the bottom left corner, the 2nd be the bottom right, 3rd be the top right, 4th be the top left. Otherwise you get distortions and overlapping lines etc.

Awesome thank you! Will this solve the angled sides as well?

slothy89, 2025-04-07:
> Awesome thank you! Will this solve the angled sides as well?

Yes, as the image of the quarter circles are causing the offset skew in the draw code.

Although, you do have a 1mm variance in one of your positions, so it technically will still be slightly skewed ;)

[image]

living-relation, 2025-04-07:
Haha thanks!