OrcaSlicer/OrcaSlicer #9199: Crashing as trying to login to Makerworld

githubuser853, 2025-04-03:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0

### Operating System (OS)

macOS

### OS Version

MacOS Sequoia 15.4 (24E248)

### Additional system information

M2 Macbook Air

### Printer

Bambu Lab P1S

### How to reproduce

1. Install fresh copy of Orca Slicer
2. Select North America
3. Select Bambu Lab P1S with .4 nozzle and .6 nozzle <-- this printer is already attached to my Makerworld profile and on my local network
4. Select Creality Ender 3v2 Neo with .4 nozzle <-- this printer is on my local network via Octopi
5. Accept default filaments
6. Accept default for Stealth Mode (unchecked)
7. Install Bambu network plugin (check)
8. Click ok when network plugin finishes
9. Click "Login/Register"
10. Select Google logo to use my Google Account for SSO (the crash also occurs if you select "System browser login")
11. Agree to Terms of Service
12. --- Orca Slicer Crashes Here ---

### Actual results

Orca Slicer crashes.

### Expected results

Orca Slicer finishes login and allows connection to Makerworld and my P1S.

### Project file & Debug log uploads

[OrcaSlicer-2025-04-02-210049.ips.txt](https://github.com/user-attachments/files/19578908/OrcaSlicer-2025-04-02-210049.ips.txt)

### Checklist of files to include

- [x] Log file
- [ ] Project file

Noisyfox, 2025-04-03:
Will take a look soon

githubuser853, 2025-04-03:
Quick update on something I found.  I have Google Chrome set as my default browser at the OS level.  To test further I switched that to Safari and it stopped crashing.  I also have switched it back and it kept working.

I'm not quite sure what was going on there but changing the default browser did something to fix this issue.

@Noisyfox I'll let you determine if you want to investigate further or close this issue with the workaround I described in this post.

wsantos1, 2025-04-06:
I am experiencing a similar issue with 2.3.0 - worked fine in 2.3.0-rc.  Clean opening of the app, no models loaded, go to the devices page and the application crashes.  I logged out of MakersWorld and it started working fine.
MacBook Pro Intel
Ventura 13.7.4

closed 2025-07-13 by github-actions[bot]

josefdahari, 2026-02-08:
I am actually having a similar issue.. All seems fine until i try to log in to bambu studio account. .. Its shows loading presets and then crashes.. Maybe i added a profile or something it doesnt like..  Its like this on 3 machines.. It just crashes and dissapears on Mac.. On windows machine it says unexpected error terminating