OrcaSlicer/OrcaSlicer #9200: (M486) Object labels cannot be turned off [2.3.0]

Rsoft1976, 2025-04-03:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

OrcaSlicer_Windows_Installer_V2.3.1

### Operating System (OS)

Windows

### OS Version

windows 10

### Printer

Arillery Sx1

### How to reproduce

1.Be sure to turn off "label objects" and "exclude objects"
2.Slice print
3.Check gcode for M486

### Actual results

#

[image]

### Expected results

M486 obj_name
M486 S0
M486 S -1
Shoudld be not present

### Project file & Debug log uploads

[benchy.zip](https://github.com/user-attachments/files/19581122/benchy.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Anything else?

#7775

Noisyfox, 2025-04-03:
Just curious, did this occur on 2.2?

Rsoft1976, 2025-04-09:
2.3

PR #9202 (2025-04-13): Fix `M486` still presented even with obj exclusion disabled
Fix #7775
Fix #9200
closed 2025-04-13 by Noisyfox