OrcaSlicer/OrcaSlicer #9211: Auto-arrange is sometimes nonsensical.

Jorvalt, 2025-04-03:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

Pictured is an example of a multicolor print. Both red objects are entirely one filament. Two other objects have red in them. I'm printing by object because that requires the least number of filament changes and prints in the least amount of time. For some reason, auto-arrange puts the objects like this and prints the entirely red objects first, then last. Arranging them like pic #2 would require one less filament change and take less time to print.

[image]
[image]

I'm not sure _why_ it does this, but it does this.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

Somehow fix whatever algorithm drives auto-arrange to realize that printing two objects that are entirely one color would be more beneficial than printing them first and last for some reason.

Jorvalt, 2025-04-04:
Small update I forgot: I realize that "print by object" is driven by object height. So it goes in sequence from shortest to tallest. I feel like it could be possible to make this a little more sophisticated than that especially when you really have no hope of the printhead ever colliding with anything that's already printed on the plate. In this case the tallest object happens to be the same color as the shortest object, but with how they're arranged, the printhead would never even come close to grazing an object here.

dewi-ny-je, 2025-04-07:
I also got a print where auto placement rotated the part alternatively in two directions, back and forth each time was run...

It's just a rough help, manual placement is mandatory whenever non-obvious placements are needed, or the bed needs to be filled

jbowers01, 2025-04-27:
My workaround is grouping and selecting all of the parts that are the same solid color and assembling them. Each assembly will be treated as a single model and printed by layer together and avoid the filament change issue while also giving you what you want the auto arrangement function to do.

rkarlsba, 2025-10-03:
And btw, please change this to a bug, it's not an enhancement!

dewi-ny-je, 2025-10-03:
Orca keeps minimum spacing between objects and that spacing is different on Bambu studio.

This thread is about a DIFFERENT feature request. And it's a feature request, not a bug. Open a discussion in the Discussions section to get help for your specific case. No code change needed.

sudo-nano, 2026-09-17:
Is it not worth discussing why the minimum spacing is there, and whether it warrants changing if it's causing undesired behavior?

dewi-ny-je, 2026-09-17:
> Is it not worth discussing why the minimum spacing is there, and whether it warrants changing if it's causing undesired behavior?

sure but not here. We stay on topic to keep discussions focused and efficient.