OrcaSlicer/OrcaSlicer #9223: Filament Profile Filtering for Different Printer Profiles.

bballbuddy25, 2025-04-04:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

Currently I have 2 klipper printers, that I built their printer profiles off of the myklipper preset. Even though they are 2 different printers, all of their filament presets show up together, making it hard to manage my filament settings for a specific printer. 

### Which printers will be beneficial to this feature?

Klipper

### Describe the solution you'd like

There should be a feature so that when I select a printer profile, only certain filament profiles are available. There is currently a similar solution, when I select my 3rd printer that is not based off of that preset, but with my 2 printers that are based off of the myklipper preset there is not. 

### Additional context

[image]

bballbuddy25, 2025-04-04:
As you can see in the picture, I currently have to label my filament profiles, and I have 7 profiles for 4 filaments (gonna be 8 when I tune asa for my E3)

Noisyfox, 2025-04-04:
Profile dependency options has been disabled at the moment. Will re-enable it once we get the UI done.