OrcaSlicer/OrcaSlicer #9229: Tree Support doesnt have an Infill - It NEEDS Infill for strength sometimes

QuackenAudio, 2025-04-04:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.31

### Operating System (OS)

Windows

### OS Version

win 11

### Additional system information

Ryzen 5 something or other
64 gb of memory
AMD video card

### Printer

Creality K2 Plus

### How to reproduce

Create anything that requires supports
add tree supports
make supports bigger
add infill of any kind to the print

### Actual results

tree support doesnt have anything inside them... hollow trunk, breaks easily, causes print failures

[image]

### Expected results

tree should have support just like object being printed - or there should be an option to turn on or off tree infill with pattern choices
would be even beter if it could have its own infill % as wellfor tree supports

### Project file & Debug log uploads

Project file

[tree - no infill.zip](https://github.com/user-attachments/files/19611083/tree.-.no.infill.zip)

### Checklist of files to include

- [ ] Log file
- [x] Project file

evilDave, 2025-04-09:
I have not seen the need for infill within tree supports, they definitely should be printable as just a shell. Check your line width possibly. I have also seen them under-extrude due to trying to print them too fast and exceeding the max flow-rate. Have you tried a wider line width for supports? (Making sure your max flow-rate is tuned so they don't under-extrude)

QuackenAudio, 2025-04-09:
theres no under extrusion happening
Just sometimes the supports need internal bracing, especially if they are taller prints and teh support is a sungle tree trunk

I can do this in CURA... my infill works in the supports

there is absolutely a need, and somebody has asked for it in a previous bug a long time ago
with comments supporting the request for it

QuackenAudio, 2025-04-10:
in cura

[image]

stephenfeather, 2025-04-11:
This definitely needs to be optional. 

The original idea behind trees was to get a big support impact, low filament impact, and low footprint on the bed and/or the print. The other argument is that they should be easy to remove (unless printed in OrcaSlicer :) )

I do print taller-narrow prototypes than most folks probably do. But I've never had a span that required an extraordinary level of strength to perform the support.

QuackenAudio, 2025-04-11:
optional is totally fine.....
but it needs to be an option

tree supports are great until they fail
I exxagerated it in the picture above
but even 5-10 % infill in a tree can fix support problems on taller prints where the standard hollow tree fails.

this doesnt change how easy it is to remove
my prints done in cura came off just as easy whether it was hollow or had some infill

Felix14-v2, 2025-12-28:
Just change the support style to Slim, Strong or Hybrid and configure perimeters count and infill template/density

[image]

QuackenAudio, 2025-12-28:
> Just change the support style to Slim, Strong or Hybrid and configure perimeters count and infill template/density

I dont know what setting you are using there, but that doesnt work....
doesnt matter if I select slim/strong or hybrid, it still doesnt give me infill on the support

[image]

[image]

[image]

For things that are single color, I have gone back to Cura because Orca developrs clearly dont know how to add features that actually make a difference when printing larger objects

no infill on supports is fine of you are printing things that are toys, but when you have 4 spools of filament in a print. it needs to have strong, stable supports that have infill in them

Erosion139, 2025-12-28:
> > Just change the support style to Slim, Strong or Hybrid and configure perimeters count and infill template/density
> 
> I dont know what setting you are using there, but that doesnt work.... doesnt matter if I select slim/strong or hybrid, it still doesnt give me infill on the support
> 
> <img alt="Image" width="1251" height="601" src="https://private-user-images.githubusercontent.com/51548286/530566431-d43f4045-8325-4e0a-a88d-1be62f4aaa62.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NjY5NjE1ODQsIm5iZiI6MTc2Njk2MTI4NCwicGF0aCI6Ii81MTU0ODI4Ni81MzA1NjY0MzEtZDQzZjQwNDUtODMyNS00ZTBhLWE4OGQtMWJlNjJmNGFhYTYyLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTEyMjglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUxMjI4VDIyMzQ0NFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTU5ODYyNWQ5MGI2NGU5ZTA1NjJkNDZkNDAxYzcyNDM1ZGEzNWZjYTJmMzhjZjg5Mjg0NjFhMWZhMjUxNTQ3M2QmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.DbovS08Tg1BXFF2Ndzc-J6mXTJIIXWgyTq-VQlKP_2o"> <img alt="Image" width="1268" height="607" src="https://private-user-images.githubusercontent.com/51548286/530566464-2ab0be11-097b-4c1b-ab42-6e79a32f30dd.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NjY5NjE1ODQsIm5iZiI6MTc2Njk2MTI4NCwicGF0aCI6Ii81MTU0ODI4Ni81MzA1NjY0NjQtMmFiMGJlMTEtMDk3Yi00YzFiLWFiNDItNmU3OWEzMmYzMGRkLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTEyMjglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUxMjI4VDIyMzQ0NFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTE1YjU0ZmVhNzFjZDVkNTY3MWM0ZDQ0MzM5ZTU3MGQ2ZjZkNGY1OTE1OWU4MDQ5MjE2NzU4MmUyNGFjZjMyNjImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.2ULBCksT0ftiEEOUqT-sQr6ZN0n9asU-TaEnEntRRys"> <img alt="Image" width="1271" height="628" src="https://private-user-images.githubusercontent.com/51548286/530566518-0b368426-4470-41e5-9b24-2ac3feb674d2.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NjY5NjE1ODQsIm5iZiI6MTc2Njk2MTI4NCwicGF0aCI6Ii81MTU0ODI4Ni81MzA1NjY1MTgtMGIzNjg0MjYtNDQ3MC00MWU1LTliMjQtMmFjM2ZlYjY3NGQyLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTEyMjglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUxMjI4VDIyMzQ0NFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWNlOWFiN2U0MjY3Y2Q0YzhhZTk5ODgwZjc0NTE0YTMzY2M3MGYyODNiMmViZDhjMWRiZmZmOTE5NGU1ZGM4NGEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.gnwT-xh--IMSXwEv46yQFvL1_cvjs4RqP2t8MXSXfGA">
> For things that are single color, I have gone back to Cura because Orca developrs clearly dont know how to add features that actually make a difference when printing larger objects
> 
> no infill on supports is fine of you are printing things that are toys, but when you have 4 spools of filament in a print. it needs to have strong, stable supports that have infill in them

[image]

In advanced under support change support wall loops to 1 and base pattern to anything but default.

Art-Porto, 2026-10-28:
My tree supports fell over too. These are hollow trunks with the Default base pattern, on a Creality K2 with Tree Hybrid: a trunk about 25 mm wide and well over 100 mm tall, printed as a two-wall tube.

[image]

I'm preparing a PR for this. The idea is a middle ground rather than infill everywhere:

- Default keeps the branches hollow, which is what keeps tree supports cheap and easy to remove.
- A trunk too wide to stand as a hollow shell gets infill inside its walls.
- Hollow stays as it is, for those who want every area hollow.

The width from which a trunk gets infill is still open, somewhere between about 6 and 9 mm. I'm settling it with physical print tests, and I'll post the results here with the PR.

Art-Porto, 2026-11-03:
Following up on the physical tests I mentioned: I printed a 178 mm mushroom on an Ender-3 V3 Plus in PLA. I sliced it in 2.4.2 with the stock process, so Default base pattern and 1 wall, and with Tree Hybrid. It stood on single-wall hollow trunks and printed fine. A 180 mm candy cane on hollow trunks held too. So hollow trunks don't fail on every print. Mine failed on the K2.

Without a print that shows where infill becomes necessary, I can't back a width threshold with data, so I'm not opening that PR for now.

I also corrected my earlier comment: the K2 trunk was a single-wall tube, not a two-wall one.