OrcaSlicer/OrcaSlicer #9237: Non-concentric Concentric after version 2.1.1

fieroscotty, 2025-04-05:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0

### Operating System (OS)

macOS

### OS Version

Sequoia 15.3.2

### Additional system information

Mac Mini M4, 16 GB, 256GB SSD

### Printer

Bambu Lab P1S or X1C

### How to reproduce

Select Concentric for your top layer, and slice. Then look at the tool path on layer 12.

[concentric-layer12.3mf.zip](https://github.com/user-attachments/files/19617721/concentric-layer12.3mf.zip)

### Actual results

In layer 12 of the attached .3mf, the concentric top pattern draws 4 times with lines separated a distance from each other, rather than right next to each other. This produced a very ugly top pattern (or bottom pattern) on a print. This behavior changed after version 2.1.1. I've uploaded screen recordings of the slicing of layer 12 in 2.1.1 and 2.3.0.

[orca2.3.0.zip](https://github.com/user-attachments/files/19617720/orca2.3.0.zip)

### Expected results

In Orca 2.1.1, layer 12 is sliced and prints with one line next to the other producing a smooth finish on the print.

[orca2.1.1.zip](https://github.com/user-attachments/files/19617719/orca2.1.1.zip)

### Project file & Debug log uploads

[concentric-layer12.3mf.zip](https://github.com/user-attachments/files/19617725/concentric-layer12.3mf.zip)

### Checklist of files to include

- [ ] Log file
- [x] Project file

fieroscotty, 2025-04-05:
it appears this may be already opened as a bug already. further research shows it slices incorrectly with a cube, but correctly with a cylinder.

MxBrnr, 2025-04-05:
Affects bottom layer pattern too.

asp1972, 2025-04-07:
The feature is important, it would be good to fix.

Noisyfox, 2025-04-10:
Dup #8383