OrcaSlicer/OrcaSlicer #9241: Infill does not join walls

MartinXO1, 2025-04-06:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0

### Operating System (OS)

Windows

### OS Version

Windows 11

### Printer

Bambu Lab X1C

### How to reproduce

Happens randomly, resolves randomly. Does happen after last update 2.3.0
The problem is more frequent on larger nozzle sizes. On 0.8 it is almost consistently present. 

### Actual results

https://www.reddit.com/r/FixMyPrint/comments/1jspcz8/infill_does_not_join_walls/

[image]

](url)

[image]

### Expected results

Random problem.

### Project file & Debug log uploads

[log.zip](https://github.com/user-attachments/files/19620520/log.zip)
[Project.zip](https://github.com/user-attachments/files/19620519/Project.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

### Anything else?

0.2 nozzle - never encountered this problem
0.4 nozzle - happens sometimes, resolves itself
0.6 nozzle - happens more frequently, resolves itself
0.8 nozzle - happens often to the point I had to stop using this nozzle size.

YZcat2023, 2025-04-06:
bro it's not a bug. Check your Factor K instead

MartinXO1, 2025-04-06:
> bro it's not a bug. Check your Factor K instead

Kindly read carefully my attached reddit post. I mentioned that I have calibrated filament profiles and this problem comes and goes randomly without me changing any of the parameters. I do not use auto lidar calibration either. Just manual pressure advance and flow calibration included in Orca.

igiannakas, 2025-04-07:
A few things I've noticed in your profile (ps I'm the guy responding to you on Reddit).

Disable this. Set it to 0 in both. It doesn't work well with bambu printers and multi color prints. This won't help you with this issue though but will save you future headache.

[image] 
There is a PR meant to address this but has not been merged in the code yet. https://github.com/SoftFever/OrcaSlicer/pull/7708

Secondly you've made significant changes to the print start gcode for your bambu. Revert these to the stock print start gcode to eliminate the possibility that any change there is interfering with the first layer.

Thirdly your selected filament profile is for a 0.2 mm nozzle but you're printing using a 0.4 mm nozzle. Pick a profile that is made for that nozzle size. I'm unsure whether this will make a difference but it should be avoided.

[image]

Finally your first layer infill is set to 105mm/sec. That is on the high side. Personally I recommend around the 55-60mm/sec mark for a fist layer, same as your first layer speeds. You may simply be going to fast for the layer to adhere properly.

[image]

From a slicer perspective the first layer appears to be sending the correct gcode commands. The lines are connecting to the walls and the flow rate and layer height are correct. So I'm almost certain that reverting all to stock and trying again should solve this issue for you together with reducing the first layer infill speed to 55-60mm/sec.

[image]

[image]

MartinXO1, 2025-04-07:
> Odoslané: pondelok 7. apríla 2025 10:48
> Komu: SoftFever/OrcaSlicer ***@***.***>
> Kópia: MartinXO1 ***@***.***>; Author ***@***.***>

Hello Ioannis, 

 

Thank you for such extensive support 😊

 

One of the first things I have tried was actually using the stock printer profile with original code. But it had the same issue.  After that I tried Bambu Slicer and that was working fine. 

Anyways, the start g-code basically just have commented out some parts of the original code, like the purge line, lidar calibration (since I don’t use lidar at all) and such things, to shorten out the print start as much as possible. No special changes really. 

Regarding the filament profile, I actually used the correct one for printing, no idea how did it switch for the project I sent to GitHub.

 

Just now I tried to open the same project from recent projects and it printed just fine without any issues. Same filament profile, same custom g-code, same print profile. 

So this is what I mean by the issue being completely random:

 

 

As for the speeds, I never touch this page, unless I really need to slow down some specific prints. But I can implement the suggested change into my custom preset, thanks for the advice 😊

I also set both of the fans to 0 like you suggested. 

Hoping for the implementation of the new Bambu Studio, pre-start fan implementation in the Orca 😊

 

Anyways, thanks again, hopefully I’ll figure out how to avoid this issue altogether.

 

Best regards,

Martin.

image.jpeg (view on web) <https://github.com/user-attachments/assets/a9285b6b-4d82-424a-88e0-2d5a20a63ee8> 

image.jpeg (view on web) <https://github.com/user-attachments/assets/460accb0-b762-49f8-a8cf-51ec0ec90d1c> 

image.jpeg (view on web) <https://github.com/user-attachments/assets/9c685006-fcfd-4e6d-a81c-db5183850bfd> 

image.jpeg (view on web) <https://github.com/user-attachments/assets/e686833a-1ec6-4c99-a452-a540809d6bcb> 

image.jpeg (view on web) <https://github.com/user-attachments/assets/bbe235dd-a480-4979-81be-7b357a5c5510> 

image.jpeg (view on web) <https://github.com/user-attachments/assets/a9285b6b-4d82-424a-88e0-2d5a20a63ee8> 

image.jpeg (view on web) <https://github.com/user-attachments/assets/460accb0-b762-49f8-a8cf-51ec0ec90d1c> 

image.jpeg (view on web) <https://github.com/user-attachments/assets/9c685006-fcfd-4e6d-a81c-db5183850bfd> 

image.jpeg (view on web) <https://github.com/user-attachments/assets/e686833a-1ec6-4c99-a452-a540809d6bcb> 

image.jpeg (view on web) <https://github.com/user-attachments/assets/bbe235dd-a480-4979-81be-7b357a5c5510>

closed 2025-07-14 by github-actions[bot]