OrcaSlicer/OrcaSlicer #9244: No Plate Visible

michauMiau, 2025-04-06:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0

### Operating System (OS)

Linux

### OS Version

Fedora 42 (beta)

### Additional system information

GPU: Nvidia RTX 4070 (proprietary drivers)

### Printer

Elegoo Neptune 4 (OpenNep4une)

### How to reproduce

Before i had the open source drivers (nouveau) and Orcaslicer was working fine
The issue appeared when i switched to the proprietary drivers 
The issue also happens on the nightly builds
I'm using the flatpak build (didn't try the appimage because i didn't have the required libraries

### Actual results

I don't really know how to describe this issue but the screenshots show what i mean

[image]

Slicing and everything works normally it's just not visible

[image]

### Expected results

The program working normally

### Project file & Debug log uploads

[3DBenchy.zip](https://github.com/user-attachments/files/19622537/3DBenchy.zip)

The project should contain all of my configurations i just added the benchy as a placeholder

[debug_Sun_Apr_06_18_30_08_8.log.zip](https://github.com/user-attachments/files/19622570/debug_Sun_Apr_06_18_30_08_8.log.zip)

I replaced some sensitive info in the log file with a # 
The project and the log is from a nightly version (yes the issue still happens on the regular version)

### Checklist of files to include

- [x] Log file
- [x] Project file

### Anything else?

The issue does not happen on windows (on the same computer)

FredFoo, 2025-04-07:
Had the same problem and it was caused by my old notebook's NVIDIA graphics card and pooprietary drivers. Changed to a new computer, no more NVIDIA, no more problems. Used the same Manjaro OS system disk, just put the disk of the old machine into the new machine and booted it up, let it select drivers for new hardware, started Orca and woops - got a build plate. 

Try this search https://www.google.com/search?q=orca+slicer+nvidia+problem

TomPcz, 2025-04-24:
I had the same issue on openSUSE Tumbleweed. Looking at pull request #8373, which is supposed to fix it, I found that the fix relies on the `nvidia-smi` command, which was not installed in my case. Installing `nvidia-compute-utils-G06` package resolved the issue.

mdziekon, 2025-05-13:
I've just upgraded my system to `Ubuntu 25.04` (from a very old version, still running X instead of Wayland) and encountered the same issue. Using `nvidia-driver-570` (`570.133.07-0ubuntu2`).

Turned out the "Zink override" fix (https://github.com/SoftFever/OrcaSlicer/pull/8373) could not apply the env vars for some reason. That reason turned out to be a missing `glxinfo` binary, which comes in the `mesa-utils` package (at least, for Ubuntu). Installing that package solved the issue.

Leaving that for other people, just in case other people have similar setup.

Also, it would be nice if Orca could warn us about missing dependencies like this, I assume it doesn't do that right now because it swallows the error somewhere.

arcanorca, 2025-06-01:
I just switched to Garuda Linux a week ago and have been dealing with this issue ever since. In my case [This fix](https://github.com/bambulab/BambuStudio/issues/6985#issuecomment-2909758521) from the Bambu Studio GitHub page worked for me. I ran these in the terminal for OrcaSlicer:

```
sudo cp /usr/share/glvnd/egl_vendor.d/50_mesa.json ~/
```

```
sudo flatpak override io.github.softfever.OrcaSlicer --env=__EGL_VENDOR_LIBRARY_FILENAMES=$HOME/50_mesa.json
```

arcanorca, 2025-06-01:
I just switched to Garuda Linux from Windows a week ago and have been dealing with this issue ever since. In my case [This fix](https://github.com/bambulab/BambuStudio/issues/6985#issuecomment-2909758521) from the Bambu Studio GitHub page worked for me. I ran these in the terminal for OrcaSlicer:

```
sudo cp /usr/share/glvnd/egl_vendor.d/50_mesa.json ~/
```

```
sudo flatpak override io.github.softfever.OrcaSlicer --env=__EGL_VENDOR_LIBRARY_FILENAMES=$HOME/50_mesa.json
```

closed 2025-10-07 by github-actions[bot]

AlexSwensen, 2025-10-30:
> I just switched to Garuda Linux from Windows a week ago and have been dealing with this issue ever since. In my case [This [...]

This worked for me on Fedora Workstation 43. Performance is still not _amazing_ but it's now usable!