OrcaSlicer/OrcaSlicer #9266: SEMM doesn't invoke filament change on BambuLab A1

otede, 2025-04-08:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0

### Operating System (OS)

Windows

### OS Version

10

### Printer

Bambu Lab A1 in LAN only mode

### How to reproduce

SEMM in Orca is undocumented so I followed the bits and pieces mostly from Prusa Slicer forums
https://forum.prusa3d.com/forum/prusaslicer/single-extruder-multi-matertial-tool-selection-t0-t1/

1. Create clean Orca Slicer project based on the default printer and settings profile.
2. Import 2 nested STL files as single object with multiple parts.
3. Verify Other - Print sequence: By Layer
4. Printer profile - Multimaterial : SEMM checked + Manual Filament Change checked
5. Printer profile - Machine G-code : {if layer_num > -1}M600{endif} //per https://github.com/SoftFever/OrcaSlicer/issues/2356#issuecomment-1751979339
6. Assign different filaments ([1] and [2]) for each of the STL parts: the plaque and the fill lettering
7. Both STL's are 0.4 mm high so verify layer height = 0.2 for total of 2 layers

[image]

[image]
[image]

### Actual results

Filament change to the second color is never prompted during the print job.

As soon as the slicing is previewed it's clear only one of the colors will be printed and indeed this happens when I send the job to the printer. This is despite the Filaments window clearly lists 2 filaments to be used along with separate times and amounts.

[image]

### Expected results

Filament change to the second color to be prompted by the printer, 1 color change to occur for my project per layer, so 2 color changes in total. I would expect to get a result similar to the following screenshots

[image]
[image]

### Project file & Debug log uploads

[debug_Tue_Apr_08_20_48_45_15476.log - Copy.0.zip](https://github.com/user-attachments/files/19656099/debug_Tue_Apr_08_20_48_45_15476.log.-.Copy.0.zip)

[Unnamed5b-branch2_02-Body1-Plaque projectorca 02 BLa1 default SEMM 2025-04-08.zip](https://github.com/user-attachments/files/19656102/Unnamed5b-branch2_02-Body1-Plaque.projectorca.02.BLa1.default.SEMM.2025-04-08.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

igiannakas, 2025-04-09:
Are you using the stock A1 profile or a modified version?

otede, 2025-04-09:
> Are you using the stock A1 profile or a modified version?

Thanks for chiming in! Yes, the stock profile. Since I normally use a fast startup G-Code profile I made sure that for SEMM (and the purpose of this bug report) I selected the stock profile FIRST and only then enabled the 2 SEMM-specific check marks and the filament change G-Code.

igiannakas, 2025-04-10:
Do you have an AMS for it or planning to do the changes manually? Sorry haven’t worked with manual filament changes with a Bambu printer so I’m not as familiar.

otede, 2025-04-11:
> Do you have an AMS for it or planning to do the changes manually? Sorry haven’t worked with manual filament changes with a Bambu printer so I’m not as familiar.

No AMS here. I am trying to achieve a manual change that is going to give me at least 2 colors per layer.

otede, 2025-04-12:
Here's the G-Code right where filament should change. M600 is clearly there. Also note that Orca itself denotes 'Filament change times: 2'. G-Code file and a revised project file attached.

[otede bug report SEMM 9266.zip](https://github.com/user-attachments/files/19721276/otede.bug.report.SEMM.9266.zip)

`
.
.
G1 X25.922 Y72.926 E.00472
G1 X25.61 Y72.614 E.00743
; stop printing object Unnamed5b-branch2_02-Body1-Plaque.stl id:15 copy 0
G1 E-2 F1800
; filament end gcode 
M106 P3 S0
M600
M106 S0
; MANUAL_TOOL_CHANGE T1
M104 S240 ; set nozzle temperature
; filament start gcode
M106 P3 S255
;Prevent PLA from jamming

M900 K0.02 L1000 M10 ; Override pressure advance value
; printing object Unnamed5b-branch2_02-Body2-textSolid.stl id:15 copy 0
; stop printing object, unique label id: 130
M625
; start printing object, unique label id: 120
M624 AQAAAAAAAAA=
G1 Z.6 F42000
G1 X33.447 Y69.756
M73 P93 R0
G1 Z.6
G1 Z.2
G1 E2 F1800
; FEATURE: Outer wall
; LINE_WIDTH: 0.255046
G1 F3000
G1 X33.814 Y70.534 E.01425
; LINE_WIDTH: 0.283589
G1 X33.85 Y70.577 E.00106
; LINE_WIDTH: 0.312132
G1 X33.886 Y70.62 E.00119
G1 X33.944 Y70.605 E.00126
; LINE_WIDTH: 0.278365
G1 X34.002 Y70.589 E.0011
.
.
`

igiannakas, 2025-04-13:
Perfect thx for the project file. I’ll take a look today

otede, 2025-07-13:
@igiannakas Any hints for me? Sorry, but the not wants this ticket buried.