OrcaSlicer/OrcaSlicer #9271: Automatic flow dynamics calibration missing

yzheka, 2025-04-09:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0

### Operating System (OS)

Windows

### OS Version

Win11

### Printer

Bambu Lab A1

### How to reproduce

1. Go to Calibration tab
2. click on Flow dynamics option

### Actual results

Automatic Calibration button is missing in flow dynamics calibration screen. 

[image]

### Expected results

Please add Automatic flow dynamics calibration button like on bambu studio

### Project file & Debug log uploads

Not needed as it is not related to the slicing procedure

### Checklist of files to include

- [ ] Log file
- [ ] Project file

codegain, 2025-05-19:
I would like to switch from Bambu Studio to OrcaSlicer because of the limitation from https://github.com/bambulab/BambuStudio/issues/4168 but its tricky if that function is missing.

Hopefully someone can implement it 👍

DJMitch117, 2025-11-04:
Bump, I am now seeing this on version 2.3.1

PG248, 2026-01-11:
I said you need a X1C to do BOTH automatic dynamic flow calibration and automatic flow rate calibration. Bambu added automatic flow rate calibrated later to the X1C. It is not perfect but it is much faster than manual calibration. I use it often.

Can’t speak for the A1 but has a H2D which has automatic dynamic flow calibration but not automatic flow rate calibration and automatic dynamic flow calibration was available in Orca 2.3.0.

That said both work in Bambuslicer and worked for me in Orca up to 2.3.0. The buttons to start either is missing in Orca 2.3.1. I don’t know whether it is only a glitch in the UI not showing the buttons or whether the code for automatic calibration has been removed.

yzheka, 2026-01-11:
@PG248 this makes sense 👍

yzheka, 2026-01-11:
@PG248 i had this issue for all versions, just reported that when i used 2.3.0 because I have found that there were no reports yet for this

LJKraus, 2026-04-02:
> I can't find auto calibration in v2.03.0 at all. 

Since there have been **two** subsequent releases, I don't think anyone will spend much time trying to fix a (possible) year-old problem with 2.3.0.

The Calibration **Tab** is a function of the Bambu network plugin, so it is not present for non-Bambu printers and is not an OrcaSlicer feature. OrcaSlicer downloads it from Bambu.

In Windows, Orca 2.3.2, and the 2.03.00.62 plugin, I see all three buttons in the Calibration tab when any Bambu printer is selected.

brummbrum, 2026-10-01:
i found the root cause. it is related to printer capability parsing by Orca (and also Bambu Studio) with specific printer firmware versions. Bambu changed the (additional ways) a printer can reports its calibration capabilities and newer printers come with more advanced features like A1. The slicer only exposes the switch when it thinks the printer can perform some form of flow calibration. Unfortunately this breaks in specific situations. The solution is tricky because we cannot simply revert back to the old way of parsing since newer printers have new capabilities and we don't want to create a mess just to restore the A1 switch. There is a surprisingly simple fix which leaves all the newly introduced flag handling and features intact and still restores the switch for older A1 printers. Pull request #16055