OrcaSlicer/OrcaSlicer #9279: Outer Brim only feature prior to V2.3

Biglogh, 2025-04-09:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

V2.3 has shaken my whole business due to brims being added to engravings on the bottom face of my parts.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

I understand why the "improvement" was made to add outer brims to all independent surfaces, but I would like the option in future versions to have an outer brim only on the object, not the surfaces within the object. Basically exactly how it worked prior to V2.3.

PeterTremonti, 2025-07-01:
Is this update the reason I can't turn off brims even though I have them disabled and set to 0mm width?  This is annoying and I don't like that I can't turn off the brim no matter what setting I choose.  I just started using OrcaSlicer within the past two months so I'm new to this slicer.

mytechguyri, 2026-04-05:
This should be re-classified as a BUG, not an "enhancement".  

"Outer brim only" is definitely BROKEN.   Frankly, this simply is NOT "outer brim only"   Its clearly almost the exact same behavior as inner and outer brims.   here are two example photos of the problem.   

First photo is "Outer brim only" , note how its putting brims in the debossed "benchy" section of my logo, but not in any of the text.  

The  2nd photo is "Outer and Inner brim" there is no practical difference in brim placement, other than it adds brims to the entire logo, and not just the "benchy" portion.  

 Its also creating brims in those two ovoid shaped holes, which are not by any definition "outer".   

This is simply wrong, it is not at all expected behavior, and it is clearly creating brims far too agressively with the "Outer brim only" setting, and in areas that nobody would ever consider "outer" even though "outer brims only" is the selected option. 

[image]

[image]

  If correcting this bug isn't an option, then add brim settings to the modifier part so that we can turn brims off in specific parts of a print.