OrcaSlicer/OrcaSlicer #9280: Camera stops transmitting after 2-3 seconds

tnxgam3rx, 2025-04-09:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0

### Operating System (OS)

Linux

### OS Version

Linux Mint 22.1

### Printer

X1C

### How to reproduce

1. Open Orca Slicer
2. If not open, go to the Device tab
3. Start the camera feed for the printer

### Actual results

The camera feed stops transmitting after 2-3 seconds, and requires you to pause and restart it.

### Expected results

The camera feed keeps streaming as long as you want it to run.

### Project file & Debug log uploads

[Debug Log file.zip](https://github.com/user-attachments/files/19675592/Debug.Log.file.zip)

### Checklist of files to include

- [x] Log file
- [ ] Project file

stephenfeather, 2025-04-10:
Could use more information:
Printer Information
Firmware of printer
Camera Model
Camera Stream information

tnxgam3rx, 2025-04-10:
Printer Information: BambuLab X1 Carbon
Firmware of printer: X1Plus 2.0 (base firmware 01.08.00)
Camera Model: Built-in camera
Camera Stream information: Unknown

mynoc96, 2025-04-12:
Seeing the same issue since updating Orca Slicer.

Executable: OrcaSlicer_Linux_AppImage_V2.3.0.AppImage
OS: Ubuntu 22.04.5 LTS
Software Version: Orca Slicer 2.3.0 Build 70931e5
Printer: Bambu Lab X1C Firmware 01.08.02.00
Video: 1080P
Running in LAN mode

Abe-Oddysee, 2025-05-20:
I have several PCs. The firmware on the Bambulab X1C printer doesn't matter. On Windows, the video stream works great in Orca Slicer 2.3.0 and Bambustudio. On Linux ZorinOS (Ubuntu 22.04), the video stream freezes after 3 seconds in Orca Slicer 2.3.0. I have the same problem with Linux Mint 22.1 (Ubuntu 24.04). It can't be Linux, because Bambustudio works great under Linux.

Even the command "mpv rtsps://bblp:password@192.168.178.44:322/streaming/live/1" shows me stable video playback.

This must be a bug in Orca Slicer. Whether it's Flatpak or the two appimages. I've tried older versions as well.

I don't understand why so few people have the same problem.
... I suspect it's the LAN mode, or perhaps no one is using Orca Slicer on Linux.

I also have a Sovol SV08 with Klipper, and Orca Slicer can't even connect to the video stream under Linux. It works in the browser, and under Windows too.

It's a shame, I actually wanted to switch from Windows to Linux.

mynoc96, 2025-05-28:
Also, it appears to be just the AppImage. The flatpak looks runs fine on my machine.

alaneb, 2025-06-15:
Same issue using the AppImage and on Linux.  2-3 seconds of video, then it freezes.  I don't own any windows machines so no way to test from there.

OS: Pop OS 22.04 LTS
AppImage: OrcaSlicer_Linux_AppImage_V2.3.0.AppImage
Version: 2.3.0 Build 70931e5
Printer: Bambu Lab X1C Firmware 1.08.02.00
Video: 1080p
Running in LAN mode

fgbb-s, 2025-07-29:
same issue as well

OS: Ubuntu Budgie 25.04
AppImage: OrcaSlicer_Linux_AppImage_V2.3.0.AppImage
Version: 2.3.0 Build
and
Flatapak: OrcaSlicer-Linux-flatpak_V2.3.0_x86_64.flatpak
Video: 1080p & 720p
Lan mode

Marvelicious75, 2025-09-07:
Same.

Printer: Bambu X1C
OS: Kubuntu 25.04
Apppimage: OrcaSlicer_Linux_AppImage_Ubuntu2404_V2.3.0
X1Plus Firmware: v3.1 
Base Firmware: 1.08.02.00
1080p or 720p
LAN mode

Astrusia, 2025-10-28:
Same issue here. CachyOS (arch based), installed via flatpak, 2.3.1 (originially 2.3.0, installed and updated via warehouse) both versions have the same issue, across several kernel versions.

X1C Firmaware 01.08.02.00 after a few seconds it stops. if you click the stop button, it transmits one (maybe two) frames
same with the A1 and mini, but they take longer, as the framerate is lower, which leads me to believe it is some overfull frame buffer or similar issue. (not a dev, sorry)

I used to be on bazzite 42 (around april to july or so) and had orca (i think 2.2.0) installed as appimage, it was a bit borked with multiple windows and froze regularly for several seconds when switching tabs but the video worked.

never had this issue on windows 10. 

if it helps i could make a screen rec of this

Marvelicious75, 2026-02-13:
For what it's worth, this problem still occasionally comes back for me, but I seem to be able to consistently fix it by shutting down Orca, rebooting the printer, and restarting Orca. Doesn't help if it happens mid-print, but at least I can fix it before I start another. Still not sure exactly what triggers it.

Doggo499, 2026-02-22:
Sam here:
Fedora 43 KDE Plasma
Flatpak 2.3.20-beta2
Bambulab X1C
LAN Mode

About the same behaviour as @HeavyHawkZero.

closed 2026-04-26 by SoftFever