OrcaSlicer/OrcaSlicer #9286: Pressure Advanced

MengorMarli, 2025-04-10:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

Feature Request: Layer-Specific Pressure Advance Setting
Summary:
Allow users to define Pressure Advance (PA) per layer or per layer range, just like existing controls for speed, acceleration, and fan settings.

Problem:
Currently, PA is set globally (usually in the start G-code or printer.cfg). However, first layers are typically printed much slower (e.g. 20–30 mm/s), while standard PA values are tuned for the regular print speed (e.g. 60–100 mm/s).

This mismatch leads to:

Overcompensation from PA during the first layer

Mini blobs, excess material at corners, or wavy skirts/brims

Unnecessary tweaking of global PA, which then negatively affects the rest of the print

Why it's useful:

Matches PA to actual speed → cleaner first layers

Prevents overextrusion at low speeds

Allows fine-tuned control for bridges, slow perimeters, or special geometries

Logically consistent with other layer-based settings (speed, accel, cooling)

This would be a huge step forward for people using Klipper or advanced flow control –
and a gamechanger for perfecting first layer quality.

Thanks for considering it!

### Which printers will be beneficial to this feature?

Klipper

### Describe the solution you'd like

Proposed Solution:
Add an option to set PA for specific layers, e.g.:

gcode
Kopieren
Bearbeiten
; Layer 0
SET_PRESSURE_ADVANCE ADVANCE=0.030

; After Layer 1
SET_PRESSURE_ADVANCE ADVANCE=0.045
This can be done through:

A new section in Print Settings > Advanced > Pressure Advance per Layer

Or via custom G-code hooks, e.g. in "Before Layer Change G-code"

johnny-bit, 2025-04-10:
Isn't that covered by adaptive pressure advance? https://github.com/SoftFever/OrcaSlicer/wiki/adaptive-pressure-advance

Ataraxia-Mechanica, 2025-04-11:
Or non-linear pressure advance? https://github.com/CarVac/kalico/blob/main/docs/Nonlinear_Pressure_Advance.md
This one works even better but even harder to calibrate.