OrcaSlicer/OrcaSlicer #9293: Signed Executable

skingd, 2025-04-11:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

Please sign the installer/executable. This will help mitigate the bad installers that attackers are pushing to the top of search.

[image]

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

Executable/installed signed with cert

yzheka, 2025-04-11:
No issue for me. Windows 11 Pro, 24H2

johnny-bit, 2025-04-11:
It's a problem - The code signing certificates usually cost more than $200 per year, there's apparently a Open Source Code Signing by Certum that's cheaper but still not free... Regardless of cost there are validation procedures that are quite involved (requiring proof of identity, website for project etc) and then any technicalities involved in signing executables (including installer)