OrcaSlicer/OrcaSlicer #9296: The configuration may be generated by a newer version of OrcaSlicer...

vordenken, 2025-04-11:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0

### Operating System (OS)

macOS

### OS Version

macOS 15

### Additional system information

The configuration may be generated by a newer version of OrcaSlicer. Some values have been replaced. Please check them:
"ensure_vertical_shell_thickness" (Ensure vertical shell thickness)
: "enabled" was replaced with "ensure_all" (All)

### Printer

Bambu Lab P1S

### How to reproduce

1. Use "open with Bambustudio" on maker world (I used this for example https://makerworld.com/de/models/1261901-minimalistic-bunny)
2. OrcaSlicer opens with said warning

### Actual results

[image]

### Expected results

The warning shouldn't be there?

### Project file & Debug log uploads

[Minimalistic+bunny+0.20+small.3mf.zip](https://github.com/user-attachments/files/19701754/Minimalistic%2Bbunny%2B0.20%2Bsmall.3mf.zip)

Log Files: there is no log file from today...?

### Checklist of files to include

- [x] Log file
- [x] Project file

ilium007, 2025-09-06:
Same issue - why was this bug closed?

[image]

wwian, 2025-09-06:
I just got this pop-up today after opening a model from MakerWorld.

Write, 2025-09-13:
Same here for all project from MakerWorld, any reason why this has been closed ?

manic101, 2025-11-10:
this is the same issue for me on windows

jermoo, 2025-12-30:
I found that I had downloaded the file for a different printer than what I have. Re-downloading the correct file for my specific printer and opening it did NOT throw the error.