OrcaSlicer/OrcaSlicer #9297: Crash when launching from mapped Network Drive (windows ver)

istyszy, 2025-04-11:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0

### Operating System (OS)

Windows

### OS Version

windows 11 

### Printer

Ender 3 MAX / PRO / S1 (all running klipper)

### How to reproduce

create a network drive ex. N:
copy portable version to N:\OrcaSlicer_V2.3.0\*
Create directory to store profiles: N:\OrcaSlicer_V2.3.0\data_dir
launch orca-slicer.exe

### Actual results

slicer crashes  with "Unhandled unknown exception; terminating the application"

[image]

### Expected results

Launch portable version of slicer with profile dirs relative to exe path.
if i copy  N:\OrcaSlicer_V2.3.0\* to local drive ex. D:\ the slicer launches with no issues.
and it loads all settings from \data_dir

### Project file & Debug log uploads

have none

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Anything else?

none,

istyszy, 2025-04-30:
nothing?

Tourniquet63, 2025-04-30:
I have the same issue.